stawel/cheali-charger #84: NiXX: ignore -dV at first 5-10 minutes of charging

stawel, 2015-03-03:


drbogger, 2015-03-31:
I like this idea! It will help prevent false positives.

bncache, 2016-01-04:
I haven't read the code yet, so please forgive me if i put my foot wrong here.
Batteries should not give any negative dV at any time during charging unless charge current is low and batteries are getting warmer while charging. Warmer batteries have less resistance than cold. This is what causes the drop in voltage when batteries are full.
Ignoring the dV for 5 minutes may be catastrophic if charger is used to top of or warm up a fully charged battery at high charge currents. 2C and up is what we used when racing 1/10 cars. Since hot batteries give more power than cold ones we popped them in the charger just before a race to top them off and get them warmer.

diggit, 2016-01-04:
This delay is optional. You don't have to worry. I need this option for some batteries. Mostly those sporadically charged (9V 200mAh cells and similar ones). I don't know why, they just start weird. IMO in case of warming up batteries when they are already full, -dV method for stopping could not work every time.

bncache, 2016-01-04:
Its risky if you don't know your charger well. Agreed :) And if batteries are hot when you start topping them off you might not get the -dV.
The resistance in the batteries decrease as the batteries get hotter. That is what causes the -dV when they are full. All the current is then converted to heat, and warms the battery, dropping the resistance and the consequent voltage.

stawel, 2016-01-05:
actually we do set a 3 minutes "ignore dV" by default, 
maybe we should set this to 0 minutes by default
(although this method is often used by other chargers because even a banana plug may
 change its resistance when newly plugged in and a current starts flowing through it)

>The resistance in the batteries decrease as the batteries get hotter. That is what causes the -dV when they are full.

This is a fact that I always wonted to check, maybe You did this?