stawel/cheali-charger #88: balance port inputs out of order usage

stawel, 2015-03-09:


stawel, 2015-07-03:
the idea is to use any balance port as the voltage input, 
example: you have a 1 cell battery and would like to connect it to the 3-th balance port
(3-th balance port is more accurate)

diggit, 2015-07-15:
Just passing around...
Most of the chargers have differential amplifiers connected on each cell and are pretty identical.
"(3-th balance port is more accurate)" - how do you know?

stawel, 2015-07-15:
Hi Patrik, 

>Most of the chargers have differential amplifiers ...

most of the 50W chargers don't have an op-amp on two first cells (on 3-inputs: GND, 1-cell, 2-cell)
we subtract the measured values in code to get the right voltages.

btw.
how is your charger-osfw project going?

diggit, 2015-07-15:
Oh  I see, this feature could be interesting.
If unused lower balancer port reports 0 when higher port is used, port to cell assignment could be done automatically.

Well I stopped charger-osfw project when I discovered cheali charger. This is practically finished FW. Now I use charger-osfw as skeleton for dumb power supply for LED lights on my ceiling, until proper power supply arrives, then I'm gonna flash it with cheali charger FW.
I rather spent my time on writing FW for HITEC Aggressor SRX with modified display (3310LCD) which is nearly finished. Soon on github.

stawel, 2015-07-15:
> Well I stopped charger-osfw project when I discovered cheali charger ...

 That's sad to hear, more diversity is always a good thing

> Now I use charger-osfw as skeleton for dumb power supply for LED lights on my ceiling, until proper power supply arrives, then I'm gonna flash it with cheali charger FW.

we actually already have  "poor's man" LED support (you can change the LEDs current,
and set a voltage cut off limit) it's in the V1.10 branch but it is not released yet,
unfortunately  I'm not sure if we ever release a full functional power supply  (where you could set also the voltage) - we will see.

>I rather spent my time on writing FW for HITEC Aggressor SRX with modified display (3310LCD) which is nearly finished. Soon on github.

great! more open source software :)

closed 2015-10-31 by stawel