stawel/cheali-charger #101: Pb (lead-acid) desulfation algoritm

Abyss777, 2015-05-08:
It would be nice to implement desulfation algorithm for Pb batteries.
There are two popular algorithms:

1. High voltage impulses (popular in English internet)

http://en.wikipedia.org/wiki/Lead%E2%80%93acid_battery#Sulfation_and_desulfation
http://www.reuk.co.uk/Battery-Desulfation.htm

I don't think it is possible to implement with our hardware, impulses should be high 20-30 V and frequency also should be high.

But I've come across with positive experiment of variation of such strategy:
   - 30 min of high voltage charging (not 6P - 12V but 7P - 14V) using Imax B6
   - 30 min rest
   - 40 min of high voltage charging
   - 40 min rest

and so on until electrolyte relative density become normal.

The minus is that user should control electrolyte relative density and other parameter of battery during charging.

2. Asymmetric charging (popular in Russian internet)

http://www.ntpo.com/izobreteniya-rossiyskoy-federacii/elektroenergetika/himicheskie-istochniki-toka/17409-sposob-zaryada-svincovogo-akkumulyatora.html

The battery is charging by impulses and discharging in pauses.

Effective frequency < 50 Hz
Period of charging <= 1/4 of whole period, period of discharging is >= 3/4 of period.

Current of charging is usual - 0.1C, current of discharging is 0,1-0,4 of charging. The most popular ratio  is 10:1 ( i.e. 5 A charging and 0,5 A discharging)

This algorithm implemented in large amount of hardware chargers for car batteries in Russian. Even in simplified variant. If charger can only charge you can select impulse charging and connect 5 W lamp parallel to battery for discharging between impulses.

It would be nice to implement second algorithm according resources of our hardware. (what is real possible frequency of such switches in Imax B6 ?)

weedy, 2015-05-13:
There is a thread on the google group/mailing list thing about this.
If I remember right the basic B6/clones can do 25-30v max at a freq of multiple kHz. I think the limit was the filter capacitors.

If we combine the 2 we could pulse 25v@50Hz + discharging for the best chance of desulfation.

stawel, 2015-05-13:
Hi guys,
I must admit that I don't have the time to work on a "desulfation" procedure,
additionally I'm not even sure whether these chargers  are capable to desulfate Pb 
batteries, but Djok39  https://github.com/Djok39/cheali-charger
is working on a pulse charge program, maybe you should ask there.

Best Regards,
Paweł

Djok39, 2015-05-18:
Hi, I found that the pulse method has little positive impact on the AGM batteries, and can not revive the dead AGM battery, so lost interest in development, but you can try to continue to improve the method. Maybe it will work on your car serviced acid batteries.

- Is it possible to increase frequency of impulses (ie up to 50 Hz)
No, it's too fast, and probably useless. The frequency should be related to the physical movement of the electrolyte.
- Is it possible to modify your mod to discharge in pauses?
possible

closed 2015-07-03 by stawel