stawel/cheali-charger #104: 200W and 400W charger audio not loud.

njozsef, 2015-05-19:
Hi Pawel!

long time a go you sended for me part  of c++ source. how to loudness the 200 or 400W

you  suggested halve the value on function: soundinterrupt: (timerperiod is half)

**cheali-charger-test1/src/hardware/200W-generic/GTPowerA6-10.cpp**

**old**
void hardware::soundInterrupt()
{
    static uint8_t on = 0;
        uint8_t f = 0;
    if(sound_ > 0) {
        on++;
    } else {
        on = 0;
    }
   **if(sound_ >= 10) f=4; //in imaxB6: f=8**
   **if(sound_ >= 20) f=2; //in imaxB6:f=4**
   **if(sound_ >= 30) f=1; //in imaxB6:f=2**
        IO::digitalWrite(BUZZER_PIN, on&f);
}

actually 1.00 version is empty:
**new**
void hardware::soundInterrupt()
{
}

You rewrite later on setBuzzer: function?

Thank you

Jozsef

stawel, 2015-05-19:
hm... that's strange,
I don't remember removing this code,

ok, I will restore it as soon as possible

Best Regards,
Paweł

GGyuri, 2015-06-30:
Hi Pawel
WHEN?
Gyuri

stawel, 2015-07-03:
should be fixed now

Best Regards,
Paweł

closed 2015-07-03 by stawel