stawel/cheali-charger #105: Firmware GensAce Imars II

gyanyeugen, 2015-05-24:
Hello.
I have an GensAce Imars II charger and suddenly started showing on LCD only black squares.
I tried to reflash with different firmware but only I have achieved was continuous beep or with another firmware intermitent beep.
I also tried to rewrite the original firmware, but nothing has changed.
Photos with my charger are located at http://www.rcgroups.com/forums/showthread.php?t=1951734&page=22
Is there any solution to fix it?

Best wishes!

stawel, 2015-05-24:
please see:
https://github.com/stawel/cheali-charger#troubleshooting
"reset atmega32 fuses using avrdude:"
also check the pin layout - which pins are connected to the LCD:
see:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/generic/50W/imaxB6-pins.h#L45-L50

gyanyeugen, 2015-05-25:
Hello Pawel.Thanks very much.
I do not know if it helps, but I looked at PCB, took some mesurements and I drew the pinout in the picture attachment file.Pins 23 and 24 are conected at 2 smd transistors under the LCD (I must desolder LCD to take masurements and a photo) 

   

https://github.com/stawel/cheali-charger#troubleshooting
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/generic/50W/imaxB6-pins.h#L45-L50—

gyanyeugen, 2015-05-26:
Thank you Pawel, My mistake, I've replied from yahoo where I put an attachment file.
The photo with pinout is located at. It seems to be another hardware architechture.
http://www.rcgroups.com/forums/showthread.php?t=1951734&page=22#post31698837

The pinouts are:
Atmega 32A        LCD
Pin 40                  Pin7
Pin 41                  Pin8
Pin42                   Pin9
Pin43                   Pin10

stawel, 2015-05-26:
>It seems to be another hardware architechture.

Yes, you are right, if you what you can try to add support to your charger into cheali-charger,
but you should be aware that this will take time.
Unfortunately I'm also a little busy right now so you will be mostly on your own.

if you are interested then you should also check other LCD connections, see:
http://www.engineersgarage.com/electronic-components/16x2-lcd-module-datasheet

It seems to me that your charger uses LCD pins:
Register Select
Read/write
Enable
DB0
DB1
DB2
DB3
DB4
DB5
DB6
DB7

stawel, 2015-07-03:
Hi @gyanyeugen,
How is your reverse engineering going?
Do You need any help?

gyanyeugen, 2015-07-04:
Hello Pawel. I didn't do  anything more to the charger. I wrote to GensAce to send me the firmware, but no response from them.I finished my 4 axis CNC hot wire and I was busy.As you said before, it takes a lot of time programming and I am not too good at that.Thanks a lot, keeping touch.
Sincerely, Eugen

stawel, 2015-07-11:
> As you said before, it takes a lot of time programming and I am not too good at that.Thanks a lot, keeping touch.

If you have free time and want to, we could try to add your charger to the project together,
I will do the programming part and you will do the testing,
what do you think?

gyanyeugen, 2015-07-13:
It's ok Pawel.Tomorow I'll go on holiday, and I'll be back after 12 days. After that we could try work on this project.
Best wishesEugen