stawel/cheali-charger #106: New Turnigy accucell 6 80W

aequitas, 2015-05-30:
I just bought the new accucell 80W (with XT60 input connector and usb port), http://www.hobbyking.com/hobbyking/store/__64345__Turnigy_Accucel_6_80W_10A_Balancer_Charger_LiHV_Capable.html

After successfully running cheali on a clone iMax B6 I am so fond of it I want it on the turnigy as well.

Does anyone have experience with running cheali on this platform? As far as I could see there is no ISP interface. There is room for a 5 pin header at the top, it could be an ISP (haven't traced the lines yet), but is missing one pin. It does seem to have Vcc and Gnd.

So far i had no luck connecting to the USB port, since the device won't be recognized under OSX:

    kernel[0]: USBF:    62715.168    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14000000)
    kernel[0]: USBF:    62715.168    The IOUSBFamily was not able to enumerate a device.

[image]

aequitas, 2015-05-30:
When starting the device in update mode (hold '-' and 'enter' during start) it does seem to register with the os as a 'Megawin Technologies' MA108 HID device (http://www.megawin.com.tw/zh-tw/product/productDetail/MA108), product id: 0x030b, vendor id: 0x0e6a.

stawel, 2015-06-11:
If a megawin CPU is the main processor we probably will not support your charger,
the CPUs architecture is just too old, but it may only be used for USB communication,
in this case you should check what is under the LCD.

Best Regards,
Paweł

aequitas, 2015-06-11:
@stawel i will see if i can get the lcd off to see whats under there.

aequitas, 2015-06-14:
It's a megawin under there, MA84G564AD4, so no luck :(

[image]

schavkin, 2016-01-01:
Hobbyking accucell 6 50W LiHV has same MPU as 80W version.

stawel, 2016-01-02:
please let me know, 
if someone of you would like to add this MPU to the project,
(unfortunately  this will cost you a lot of free time)

NIXIE-123, 2016-03-23:
Hi!
I have accucel 6 50W 6A lihv. And have the same problem. There is no option to calibrate. That's pitty, because balancer is inaccurate. I've read, that is possible to change firmware via usb. I'm waiting if (hopefully) someone will work on this, because those chargers are getting popular.
Best regards,
Pozdrawiam

Mikimimi, 2016-05-05:
Hi,
i have Imax b6 AC V2 that also has megawin MA84G564. It is genuine SkyRC. The charger is off by 0,4V. I would like cheali on this charger too.

Here is the picture of the board [https://lh6.googleusercontent.com/proxy/QrsrvHqYg-u1As7IhC4pMT4c-nvTGiDhdVjL-qzNXh1YDWbDJpY7Qd9zsuP4kazvK4sp0C8hTwRV07Q_fFHD1d0XvhE4r-zSOyFXOePq=w5000-h5000](url)
Here is picture of pinout [https://lh6.googleusercontent.com/proxy/9RIRwpSYj6fNPvq7Gvg4qfi48pbyDKaOOeQoC_Lehb-YNSPRpu9Y_7VDC2BnqgIdUD598tU08sZeI0Yw1lomJZR39y1Re5ehJKZkOlP6AFJtbEo=w5000-h5000](url)

Here are videos:
[https://www.youtube.com/watch?v=cPoWtJsmpwo](url)
[https://www.youtube.com/watch?v=ZhkbdxS_noY](url)
[https://www.youtube.com/watch?v=FJisTvgV2mk](url)

stawel, 2016-05-30:
Hi guys,
I have bad news for you,
I've decided to drop the attempts to add support for the mcs-51 CPUs,
reason: currently the cheali-charger architecture is 
to complex for such a simple CPU, also the open source C compile ssdc 
has very poor optimization capabilities.

All my test code is stored in this branch:
https://github.com/stawel/cheali-charger/tree/mcs-51  
I managed to:
1. implement a simple  C++ to C translator (ssdc support only C)
2. rewrite the code so it can be complied with ssdc 
3. reduce the RAM usage to ~750B
I also:
4. disabled all possible features (including calibration)
5. added an empty mcs-51 hardware implementation (all hardware specific methods are empty)

After all of that  I was able to compile the code but  the code size was about 35kB.
Looking at the disassembled code it is clear that the ssdc compiler
is not suitable for such a big project - unfortunately we grew too much
(it's probably not the compilers fault, the CPU's architecture is just too simple)

nevertheless It looks like it should be possible to run cheali-charger on MA84G564,
but I hoped to go down with the code size below 32kB so it could be run also
on other mcs-51 CPUs.
Given that I had to almost entirely rewrite the cheali-charger code (point 1. and 2.) and the only 
benefit would be the support for one charger, I've decided to drop these efforts.

Sorry guys,

Best Regards,
Paweł

aequitas, 2016-05-30:
Hi Paweł,

thank you very much for the effort you put into this! To bad it doesn't work out for us, but it's understandable. 

grtz,

Johan

LTMNO, 2018-09-05:
[image]

Dooh... i have an old original.... ;-(
Is the problem the architecture of the code(upgrade) with this chip.. or the updating process?

Baterka, 2022-01-20:
Accucel 6 80W HW version 1.00 still not supported? I really hate the min. 3.0V cutoff voltage for LiPo DISCHARGE... Many today's cells are 2.5V

diggit, 2022-01-20:
If it has same mcu as in photo above, then it's not supported.

diggit, 2022-01-20:
Typical Lixx cells have minimal amount of energy below 3.0V. It's not worth to discharge cells so deeply.

Baterka, 2022-01-27:
> Typical Lixx cells have minimal amount of energy below 3.0V. It's not worth to discharge cells so deeply.

Not really agree. Today's 18650 are rated to 2.5V safe discharge...

diggit, 2022-01-27:
Yes they have 2.5V cutoff. I am just saying, it's not worth to discharge cell so deeply. Eg. Sony VTC5A (2.0 V cutoff) has less than 4% SOC at 3V and close to 0% SOC at 2.8V.
There are probably cells with different characteristics.
But we are getting OT here.

ossilator, 2026-09-28:
so, it's a decade later, and i wonder what the state of the art is.

does SDCC still suck so much at optimization that it's unrealistic to use it? somebody feels like re-testing stawel's branch?

rewriting in c might be unnecessary, as there is now a [C backend for LLVM](https://github.com/JuliaHubOSS/llvm-cbe). it shouldn't be too hard to integrate this into the build system. however, it's plausible that the c++ code would want to use target-specific language extensions which the clang-based translation would fail to support.

for the flash-programming side, there is [a project](https://gitlab.com/hb9xar/z51dude) that targets abov's OnChipDebug.