stawel/cheali-charger #108: How about control input (for example serial commands) ?

AndKe, 2015-06-20:
Most of these chargers does output a serial stream for monitoring.
I would like to be able to control them over serial protocol.
Like: select Li-po , current, cell count, maybe maximum time and max capacity , then "start" or "stop"  

What do you say ? - doable ? is the serial RX pin unused on most chargers and just need to be connected ?

I would do programming if it's viable.

stawel, 2015-06-20:
>What do you say ? - doable ?

it's a great idea, in fact many people were asking for it.
 
> is the serial RX pin unused on most chargers and just need to be connected ?

depends, if you are talking about the arm M0517 cpu then there are 2 free UARTs available,
but on the atmega32 version the are no free pins:
- 200W chargers: use RX to control the fan
- 50W chargers:  use RX for balancing (cell 4)

you could also swap some pins, but you have to sacrifice something

specing, 2018-05-03:
There is another option: bitbang i2c or some other bidirectional communication protocol and make the clock async. Use FX2* (<$2.5) as usb-i2c.