stawel/cheali-charger #114: Discharge problem in 400W version

GGyuri, 2015-06-30:
Hi Pawel

Discharge program woking well in 8150 and 200W version. 400W litle problem:

I setting 6S 1200mA start program and current max 818mA
I setting 4S 2000mA start program and current max 818mA
I setting 2S 4000mA start program and current max 818mA

The another charger (diferent the calibration data)

I setting 6S 1200mA start program and current max 1080mA 
I setting 4S 2000mA start program and current max 1080mA
I setting 2S 4000mA start program and current max 1080mA

Please check it the discharge program in 400W version

Thanks
Gyuri

GGyuri, 2015-06-30:
Hi Pawel

I'm sorry but I forgot.
Please modify the dishcarge power max from 25W to 40W the 400w version.
Safe limits in 8150 and 200W versions: 30Watts (max35W) 400W and 2X400W version: 40Watts
I tested the limit data last summer.
50Watt lhe lot!

THX
Gyuri

stawel, 2015-07-03:
ok, I've fixed the bug with too low discharge current, please check if it's ok now,

but I didn't change the maximum discharge power because of:
Maximum discharge capacity: 25W
in: http://www.hobbyking.com/hobbyking/store/__17422__TURNIGY_MEGA_400W_V2_Lithium_Polymer_Battery_Charger_Version_2_.html

closed 2015-07-11 by stawel