stawel/cheali-charger #119: 400W Fan problem while charging

adfolfotregosa, 2015-07-07:
Hi.
Unfortunately the fan problem is not solved while charging on the 400W version.
Please see youtube video. It seams to work correctly when you stop program ??

https://www.youtube.com/watch?v=Bdglm39yd5I

(please ignore the backgound fan noise, it comes from power supply for the charger)
Thank you

stawel, 2015-07-08:
Thanks for the video!

is this firmware the latest one? with the date: 20150704

did you calibrate the internal temperature recently?
or has your eeprom.eep file your current calibration on it?

could you also do a second test/video for me?
I will send you a new test firmware and I'm interested 
in the "Tint=" reading you should me during this video,
(actually it would  be great if you could reproduce the whole video)

Best Regards,
Paweł

adfolfotregosa, 2015-07-08:
Yes , 20150704.
I did on previous firmware, not on version 20150704. eeprom I sent you has stock calibration values. I did not find need for it. It was measuring correctly enough. 
Yeah sure. Please explain exactly what you need to see and I'll give it a try.

Thank you

stawel, 2015-07-08:
> Yeah sure. Please explain exactly what you need to see and I'll give it a try.

Thanks,
when you start the  "charge" program and go to the temperature screen, you will see
a "test=" reading, I would like to see the values of it when your fan 
starts to switch on-off.

adfolfotregosa, 2015-07-08:
Ok. I hope it shows you what you need. I did a jump on the video to exactly the time were the fans kick in.

https://www.youtube.com/watch?v=QT8V7kWCH0w

stawel, 2015-07-08:
Thanks, this is exactly what I needed!

the "test=" value is a "pure" ADC temperature reading,
and the "Tint=" is a averaged value of 1000 such samples,
unfortunately we turn fan on-off based on this "pure" ADC reading
and in your case it's not very stable.
Taking into account your calibration, a reading of:
274 is equal to 45C
316 is equal to 40C
and in your video both values were exceeded.

This bug requires a lot of code rewriting,
so unfortunately I wont  fix it until the next major release.

Best Regards,
Paweł

adfolfotregosa, 2015-07-08:
No problem. I'll make the fans stay on during charging etc etc

Thank you

stawel, 2015-10-31:
should be fixed now (actually since 11 jul), please reopen this issue if it's not

closed 2015-10-31 by stawel