stawel/cheali-charger #125: Deviation for serial output interval

shadow974, 2015-07-31:
Im trying to build a log parser based on a ESP2866 which pushes the data to a webserver/database.

Im using the HobbyKing Quattro Charger which is four times an ImaxB6. I have to cummulate 4x serial data with an own processor, like an arduino with 4x softserial. The problem is, the four charger are sending their log output at the same time, which overloads the softserial on my arduino.

If the interval could be deviated by some ms +/-, this promblem could be solved.

Any other ideas?

stawel, 2015-07-31:
do you mean intervals between two bytes or two transmissions?
nevertheless I'm not sure if this would help you in any way, 
since each transmission is started independently and they can shift 
a little bit in time.

other ideas :-)

1. write your own  softserial library, the arduino softserial is not design for such heavy-duty work,
  the avr processor allows you to read logic states of 8 pins at the same time,
  this should be enough for your needs
2. another solution would be to add serial commands to cheali-charger (RX communication), 
    some people asked for it  already, but for that you need to somehow 
   gain an extra CPU pin in your charger  (RX pin is occupied),
   and  you would have to write the required code yourself. 
3. you could synchronize all transmissions in time based on power up time,
   and the shift each transmission accordingly to your needs (although I'm not sure if this will work)

shadow974, 2015-08-03:
ok, have to check this agin. I thought that the Timer is started on powerup and the transmittsions are synchronized to this timer.
I think you can close this ticket :)

stawel, 2015-08-03:
I may be wrong, but it should be synchronized to the "entering"  into a program ("charge", "discharge" and so on).

closed 2015-08-14 by stawel