stawel/cheali-charger #126: "new" charger support

diggit, 2015-08-03:
I bought "TURNIGY ACCUCELL 6" charger. Black text on blue display was first surprise. Almost immediately, I've disassembled it. Another surprise, different PCB layout. Main IC is completely blank o top (48 pins). But VCC, XTAL, AINs seems to be matching nuvoton's ARM pinout.
* LCD seems to be connected via 8bit BUS.
* FAN is switched by separate GPIO. 
* balancer load is composed just by 2 parallel resistors instead of 2 (same package)
* balacer measure resistors for OP AMP are 1%
* rectifying diode is in TO-220, using rear side of case as heatsink

Unfortunately nuvoton's datasheet does not talk about JTAG and there is no programming header (pin holes below buttons are connected to buttons and coupled with ground by capacitors)
I have FT2232 which supports JTAG.

PCB front
[image]
PCB back
[image]

stawel, 2015-08-03:
hm.. 
we had a discussion about this CPU  (not sure if it's exactly this one) 
on out mailing list and nobody could figure out what kind of CPU this is
(there are also some threads on rcgroups:
http://www.rcgroups.com/forums/showthread.php?t=1046318&page=11 )

the nuvoton m0517 cpu has:
1. (+) nRST on pin 4 (reset) maybe this one too.
2. (-)  LDO_CAP on pin 18 (should be a capacitor on this pin) 
3. (-) AVss on pin 6
4. ICE_DATA pin 31 - (Serial Wire Debug)
5. ICE_CLK pin 30   - (Serial Wire Debug)

you could test if it's a nuvoton, by connecting pin 30,pin 31, pin 4, 
to a JTAG but the JTAG must support "Serial Wire Debug"
(there are some cheap stm32 jtag programmers on ebay).

But my guess is it's some king of a mcs-51 CPU, and the programming pins are TP1-3.

diggit, 2015-08-03:
Oh, what a disappointment. LDO_CAP pin is used as GPIO, so no chance of already supported nuvoton MCU. I quite like PCB layout, maybe I'll try some nasty MCU replacement.

closed 2015-10-31 by stawel