stawel/cheali-charger #129: Battery-ID as an selectable option which is then sent in the serial-log

shadow974, 2015-08-05:
Hi,

i'm building a logging-device for our chargers and had an idea for better logging possibility.
I planed to put every serial-log-line into an SQL Database.

All my batteries have uniqe numbers printed on them. My idea is to get these numbers in my database. 

This could be a variable in cheali-charger that can be entered/edited before every charge or programm-select. This value can be than transmitted in the serial-log and so on its way to my database.

What do you think?

best regards,
 Karl

stawel, 2015-08-09:
Interesting idea,
but you have to take into account that we are running out of eeprom memory and
a unique number may not fit into it (multiplied by the number of stored batteries) , depends how long it will be, if it's a 16bit  integer there shouldn't be any problems.

alternatively you could send just the batteries index and do some mapping on the other side,
also a good idea would be to send a unique charger ID configurable in settings.

shadow974, 2015-08-10:
my idea was to provide a selection menu just between the "select programm" and "start programm". It should be hust a nummerical selection between 1 and 255 for example. I think nobody has more than 255 batteries at home :) The mapping is completly done on the receiving side.

So for example select "3S 2200mAh" -> "Charge/Balance" -> "Battery ID" -> Charge start. This menu could also be disabled in the options for users without such a logging-device. Or just included with a compiler-flag..

The unique "charger-id" is generated by the transmitting part connected to the serial-port of the charger.

stawel, 2015-08-12:
hm... sounds ok to me, assuming it can be enabled/disabled during compilation

just thinking, also a good idea would be if you could use a scanner device to select a battery
for example a IR scanner or a  phone  QR code scanner.

shadow974, 2015-08-12:
This would be perfect, but we don't have an easy way of inbound communication. UART-RXD is already used for other tasks.

davidrib, 2015-08-12:
would be possible the charger read a 1-wire memory/ID_chip with charge parameters to be fully auto???

stawel, 2015-08-13:
> This would be perfect, but we don't have an easy way of inbound communication. UART-RXD is already used for other tasks.

you are absolutely right, but there are some "hard" ways to do this :)
the question is, do we really  want an inbound communication,  if yes, who will implement it and which solution is the simplest one,
our options are:
1. use BUZZER_PIN to control BALANCER4_LOAD_PIN (BALANCER4_LOAD_PIN - is currently occupying UART'S RX pin) , downside: no sound
2.  use one of the LCD pins to also control BALANCER4_LOAD_PIN, downside: cell4 will be discharged for some microseconds when display is refreshed  
3. use one wire communication through TX pin - I'm not sure if this is implementable, although we have a unused timer
4. connect all buttons through a resistor ladder to A5 (BALANCER3_LOAD_PIN) and use the free pins for BALANCER3_LOAD_PIN and BALANCER4_LOAD_PIN, downside: complicated
5. .... probably many more ....

stawel, 2015-08-14:
Another benefit: if we would have also incoming communication you could connect all charger to one  UART (parallel)

shadow974, 2015-08-14:
Dont need parallel UART anymore. My ESP8266-Chip is capable of 4x Software-UART at the same time :)

shadow974, 2016-02-12:
Ok, it seems that C++ is a little bit too abstract for me. Have no idea where to start implementing the "Battery ID Screen".

stawel, 2016-02-13:
hm... "Battery ID Screen", maybe you would like to store one ID per battery? (in "edit battery" menu)
in this case you need just to add one line:
https://github.com/stawel/cheali-charger/blob/master/src/core/menus/ProgramDataMenu.cpp#L91-L119

if you prefer a menu like:  "3S 2200mAh" -> "Charge/Balance" -> "Battery ID" -> Charge start.
you can change: Program::run(prog); to something like: ProgramMenus::selectBatteryId() and create the menu in it (and also run Program::run(prog) in it)
https://github.com/stawel/cheali-charger/blob/master/src/core/menus/ProgramMenus.cpp#L160

shadow974, 2016-02-13:
Perfect, thanks for your hint. I already found the Program::run(prog) part but was not sure if i was right :)