stawel/cheali-charger #133: internal resistance not shown

neoedmund, 2015-10-24:
I found "internal resistance" info are not shown anywhere.
I looked into source and find 
'''
            {Screen::Balancer::displayResistance1_3,PAGE_BALANCE_PORT, PAGE_START_INFO + PAGE_PROGRAM(Program::Balance)},
            {Screen::Balancer::displayResistance4_6,PAGE_BALANCE_PORT, PAGE_START_INFO + PAGE_PROGRAM(Program::Balance)},
BALANCER_PORTS_GT_6(
            {Screen::Balancer::displayResistance7_9,PAGE_BALANCE_PORT, PAGE_START_INFO + PAGE_PROGRAM(Program::Balance)},)

            {Screen::Methods::displayR,             PAGE_ALWAYS, PAGE_START_INFO + PAGE_PROGRAM(Program::Balance)},
'''
It seems "internal resistance" will show in some conditions.
Would you explain what PAGE_START_INFO mean and how to make it shown 
Thanks in advance.

stawel, 2015-10-24:
PAGE_START_INFO is what you see when you select a "program" (charge, discharge ...),but before starting it (before you long press the "start" button).

a line like:
{Screen::Balancer::displayResistance1_3,PAGE_BALANCE_PORT, PAGE_START_INFO + PAGE_PROGRAM(Program::Balance)}
means:
display displayResistance1_3, when:
1. PAGE_BALANCE_PORT is set (something is connected to the balance port)
but excluding:
1. PAGE_START_INFO -- program has not started yet
2. PAGE_PROGRAM(Program::Balance)  -- when the program is set to "balance" 
   (we don't have any internal resistance data when we do only balancing)

neoedmund, 2015-10-25:
Thank you, now I long press the start button to start a program and found them :)

closed 2015-10-25 by neoedmund