stawel/cheali-charger #139: Please fix 'Hardware' section - HobbyKing clone NOT a clone

sylvandb, 2015-12-10:
The link in the supported hardware section to HobbyKing clone IMax B6 links to a charger which looks the same, but which is NOT compatible: http://www.hobbyking.com/hobbyking/store/__15379__IMAX_B6_Charger_Discharger_1_6_Cells_COPY_.html

I ordered (from USA East) and received this week that "not a clone" clone charger and found it has an unmarked microcontroller with a 12MHz crystal. There does appear to be a 5 pin layout for programming. The firmware does still sign on with the imax b6 intro text. Neither of the two setup menus are accessible, instead there is a single status screen accessed by holding STOP and START while connecting power.

Anyone know with reasonable certainty of a currently available charger which is compatible with this project?

stawel, 2015-12-10:
sorry to hear that,
I removed all hobbyking links,

>Anyone know with reasonable certainty of a currently available charger which is compatible with this project?

you could ask on the cheali-charger forum, but 
you should be aware that buying even from the same seller 
doesn't gwarant you a supported charger.
There are currently many  ""not a clone" clones" on the market.

sasam, 2015-12-10:
This is lottery...

I look around and found an discount on TomTop webshop. I will try lucky
with this one:
http://www.tomtop.com/imax-b6-digital-lcd-lipo-nimh-battery-balance-charger-rm163.html

I just order one, and hope this charger is with ARM micro-controller. We
will see am i lucky or not :)

sylvandb, 2015-12-11:
Yes sasam, this is a lottery.

Good luck with your tomtop charger. I will not try my luck today, it looks too much like the one I received from hobby king.

Please post what you receive from tomtop.

sasam, 2015-12-23:
Yesterday i took my new charger from port office (it arrived 2 days ago but
i wasn't at home when my postman arrived so I must went to post-office to
take this charger).
I won an Atmel based charger, PCB look like original layout.
This is my 3. charger, and first with Atmega32 CPU. Other two charger are
with Novuton M0517LBN CPU.

This is link to this charger:
http://www.tomtop.com/imax-b6-digital-lcd-lipo-nimh-battery-balance-charger-rm163.html
When I ordered this charger there was free-shipping action, so I pay $12.09
for this charger.

Last night i play with original SW. I try to charge NiMhi battery pack, and
play with temp sensor and serial port. It seem sthat both work as expected.
Little surprise for me is serial port with binary protocol. First I tray
with standard terminal but it was unreadable. When I connect this port to
Logview, adjust Device as Imax B6, LogView start to work as expected.

What I need to flash this charger ? (which programmer board)

Best regards,
saša

sylvandb, 2015-12-23:
That is good news!  I decided to try my luck again. :)  No free shipping for me (and $8 for shipping is a shame) but must pay to play.

For a programmer, a "usbasp" like this one should do the job: http://www.banggood.com/USBASP-USBISP-3_3-5V-AVR-Downloader-Programmer-With-ATMEGA8-ATMEGA128-p-934425.html  There are others maybe nicer (in a case or with multiple types of connections) but also more expensive. Search for "usbasp" or "usbisp" or "usbtinyisp" or "atmel avr isp".

Or if you have an Arduino you could wire up your own programmer using the "Arduino ISP" sketch: https://www.arduino.cc/en/Tutorial/ArduinoISP and/or https://github.com/rsbohn/ArduinoISP

sasam, 2015-12-23:
Pay attention, that on this site you can order from different store, and
shipping costs are not the same. For example I am from Croatia and shipping
from UK is cheapest solutions.
If I good remember my ARM base chargers (M0517LBN ) I order from ebay but
from some UK TomTop store. it was cca +1.5Y & 1Y ago.

When I saw picture from your "usbasp" link i remember that when i order my
first charger I order something similar.I expect Atmel CPU, but on
"lottery" i win ARM based charger so I must find programmer board for ARM
programming. I have Bu-link, and 2 version of ST-Link-V2.  I hope that I
will find "usbasp" version.

PS.
Yesterday I send few photos of this charger in the e-mail attachment. Did
you revived this photos ?  It seems that GitHub application drop attachment
when you send it over e-mail.

Best Regards,
Saša

sylvandb, 2015-12-23:
I am in USA and I did pay a little more shipping to order from USA warehouse. I will post what I receive...

I did not see pictures with e-mail. I can see them attached directly on github.com. Those are very good pictures looking under the LCD.

sasam, 2015-12-23:
This charger arrive to me from Hong Kong (CH warehouse).
I attach pictures directly on GitHub. I take cca 30-40 photos under LCD, and only on few of them I have readable CPU label.

sylvandb, 2016-01-01:
I received two chargers from TomTop USA warehouse. I think they are Atmel.  :)

I have not been able to see the microcontroller with my cheap borescope, but the boards appear as pictured by sasam with the same empty 6-hole programming header and a 16mhz crystal.

Also I can get into second calibration menu (ENTER + DEC) on one charger, the other one I think I could but it has really bad switches...

Thank you sasam!

sasam, 2016-01-03:
> I received two chargers from TomTop USA warehouse. I think they are Atmel. :)
>

Great!
I found my USBASP programmer :) but I have not time to try, how it work. 

>  I have not been able to see the microcontroller with my cheap borescope
>

It is possible to take usable photo with mobile phone camera (under LCD display). When my first ARM charger arrived i take this photo with my Samsung SII phone.

[image]

closed 2016-07-23 by stawel

Johandersen, 2016-10-31:
Thank You so much for sharing! I bought it from German warehouse and it was Atmel
I flashed the "clone" hex and it works (didn't try charging) but it looks just like [the original board](https://github.com/stawel/cheali-charger/blob/master/docs/imaxB6/imaxB6-original-front.jpg) and flashing pinout is also like original
Should I reflash with .hex named "original"?