stawel/cheali-charger #142: support for Turnigy MEGA 400Wv2 ?

Thomas-1985, 2016-01-07:
Hi

I am wondering (i want to buy a charger with more than 5A like my Imax B6) if you support this charger already:

http://www.ebay.de/itm/RC-TURNIGY-MEGA-400W-V2-Lithium-Polymer-Battery-Charger-Version-
2-/221559383389?hash=item3395f7e55d:g:D8AAAOSwEgVWS4OG

I saw that the MEGA 400Wx2 is already supported, but i am unsure if this includes the "single" 400W charger.

Cheers

Thomy

stawel, 2016-01-07:
depends on the CPU, if you buy a atmega32 version it will work,
unfortunately there is no easy way to tell what CPU is inside your charger.

petri1973, 2016-02-24:
Hi.

I read from somewhere here that the mega 400w v2 needs 2x Schottky diodes replaced..?
Is this so or is that modification obsolete? Have a guide page where more info if this mod is needed?

Pete

petri1973, 2016-02-24:
Sorry, one schottky in v2 version...

stawel, 2016-02-26:
I'm not sure what's with this Schottky diodes, please ask this question on 
cheali-charger@googlegroups.com maybe you will gen an answer there.

Kergekoin, 2016-02-26:
Got my V2's from the HK a few days ago. They have atmega32a inside. So sould be good to go with the latest Mega 400W X2 version.
I am still waiting for my atmel USB programmer. So it takes a while until i am able to flash them and report back. As i found from google, the pins should be 1:SCK,  2:MISO,  3:MOSI,  4:RESET,  5:GND,  6:VCC on this one.

closed 2016-07-23 by stawel