stawel/cheali-charger #144: possibility to auto-charge on startup

Thomas-1985, 2016-01-11:
Hi

Just a feature suggestion from my side (very nice project by the way)

I just thought if it would be nice to define a preset (i.e. "LiFe" "charge" at "5A") and if the charger is powered up and a battery is detected, the charging starts automatically starts (best option with a configurable delay, i.e. 30 seconds).

This would make the charger and excellent device to charge i.e. 4 LiFePo4 batteries in a camping van once the engine starts.
The charger can cope with variable input voltage and charges/monitores the batteries better than just bleeding-BMS connected to each battery cell and then the whole pack attached to 12V.

If there would be a configurable re-start voltage (if the battery is drained below this level the charger re-starts) this would be even better :)

What do you think about this?

Best,

Thomy

Thomas-1985, 2016-01-31:
It would be nice if this could be implemented

sfabris, 2016-02-01:
Hi Tomy,

I'm usign Cheali charger in camping van to charge a LiFe pack.
However, automate starting is the last of the problem I've encountered so far:

- Usually camping van have a common ground. You can't use a common ground while using one of the charger supported by Cheali charger. So I have to fully disconnect the battery under charge everytime...

- Charging current: I already burnt the pcb tracks of a couple of Turnigy MEGA 400... I want about 30 amps continuous in my pack.

So, auto charging (which I think could be easily implemented) is last of my camping van priorities.

Thomas-1985, 2016-02-01:
Hi

Please implement if it can be done :)

I know about the common ground problem, but in my setup i am using a 12v 
relais to completely disconnect the battery while charging.
Additionally 5A will be fine for me (or less then i am not burning my 
Imab B6 charger).
As i said it would make cheali-charger an ideal replacement for 12v LiPo 
charging (i haven't found any alternatives so far).

As i have set my battery profiles in the charger, i just need an option 
for "start auto-charging ID X after X seconds" (with X beein adjustable).

Best,

Thomy

zpukr, 2022-12-13:
There is no solution to this problem even after 6 years?

zpukr, 2023-10-16:
I implemented a fork with auto start support, source files and ready-flash hex firmware on GitHub