stawel/cheali-charger #145: Dis-/charged Energy is NOT Charge*Voltage

johslarsen, 2016-01-20:
After discharging a battery and disconnecting it I noticed that the accumulated energy displayed 0.0Wh. This indicates that it is somehow related to the current power consumption (0W with disconnected battery), which is clearly wrong.

Examining the source (`AnalogInputs::finalizeFullVirtualMeasurement()`)shows that energy is calculated as a product of accumulated charge and voltage at this measurement, which explains the described behavior since voltage is 0 when the battery is disconnected.
Electric charge is the integrate of current, so it is a function of I(t), where (t) means instantaneous at the time of measurement. Energy, on the other hand, is an integrate of power, so it is a function of I(t)\*U(t). Today's implementation calculates it incorrectly (I(t)\*U\_now), yielding a misleading result; and it completely misses the reason for measuring accumulated energy.

Fixing this should be doable by adding similar code for accumulating power (I\*U) alongside that which accumulates charge in `AnalogInputs::doSlowInterrupt`, and adding something similar to `AnalogInputs::getCharge` for converting it to (m)Wh. If you assume a **constant** current (i.e. discharge mode) an alternative implementation would be to keep a moving voltage average during the discharge period an correctly estimate energy from electric charge and average voltage, but this would not be any simpler and would not work for dynamic current (i.e. charge mode).

stawel, 2016-01-20:
Thanks for the issue report I will fix this before the next release,
...but a simple: "energy is computed wrong"  would also do the job ;)

Of course you are absolutely right about the "energy" calculations,
if I'm not mistaken iggnus already fixed this issue in his branch:
https://github.com/Iggnus/cheali-charger-i1/blob/master/src/core/AnalogInputs.cpp#L381-L391

btw. the main problem for me is not the knowledge what energy is, but rather
how to fit it into a uint32_t variable in a convincingly way.

johslarsen, 2016-01-21:
I prefer descriptive issue titles, but in hindsight the uppercase *not* might have been a bit excessive, sorry if it was perceived condescending, that was not the intention.

Yeah, the branch you referred accumulates energy in the function I mentioned. I would prefer to do the least amount of needed work in a interrupt handler, but overall that seems to do the trick. I presume it only stores it occasionally (each 88th time), to have the voltage measurement taken at points in time that drift with respect to the charging process. Which, hopefully, distributes the relative measurement point uniformly within the charge process cycle, but I have too little knowledge about the different charging processes to understand how that impacts the accumulated value.

What would be the problem of fitting it in a uint32_t? It is not like the integer will overflow (~4kWh if storing as with uWh precision) during regular usage. The aforementioned branch seems to internally (i.e. `Eout_`) store it like a 100uWs accumulator in a uint32_t variable without any problems, and converting it to Wh when it overwrites `Eout`.

stawel, 2016-01-23:
what do you think about the commit?

johslarsen, 2016-01-25:
It is adequate, and the logic seems sound. Personally I would do more refactoring, but that is not important.

See, johslarsen@d5daf3faab3e6add1aff1f14ae3744b5d9edcaae and johslarsen@cbf133eac1e1cd73333a762f597d6f80c9277769 and johslarsen@2692a5306ff7894df1568d6b49a9f5ccb49d8a35 if you are interested in some concrete refactoring suggestions, but note that these have only been compiled and never been tested.

I could not figure out a very suitable name for the function that converts measurements taken during interrupt periods to seconds, and I am more fluent in C than C++, but I doubt static inline functions are very idiomatic C++, nevertheless my idea of not duplicating this calculation should be clear.

stawel, 2016-01-25:
I like your proposals, but I do have a few comments to them 
(I will add the comments in a moment)

also I would be honored if you could add my suggestions and
"squash"  everything  into one commit (including https://github.com/stawel/cheali-charger/commit/7ee2714bfd6a8c574435ce38243ff3a120377c05 ).
This way you would be a contributor to cheali-charger and I could blame you if something 
doesn't work ;)

of course if you don't have the time, I will do it myself.

johslarsen, 2016-01-25:
I am no fan of changing published history, but if that is what you would like, then fine by me.

The squashed commit with changes can be found at johslarsen@b00b358446cd4301bd55c10db9aaad229c2844ad, and I doubt github's pull request are fine with rewriting history. BTW, you wrote the code, so you are still the author, and I am only the commiter unless someone explicitly updates author.

stawel, 2016-01-25:
I've commented something wrong :( sorry for that, I'll add some more comments in a moment.

>  BTW, you wrote the code, so you are still the author, and I am only the commiter unless someone explicitly updates author.

"git blame" prints your name on each edited line,
is there some other method to check the author?

johslarsen, 2016-01-25:
I built and flashed these changes to my charger for testing, and I kept getting CapacityLimit errors right after starting dis-/charging process. Flashed the original version, and it work fine (i.e. problem in this changeset).

Turns out merging the division statements in `AnalogInputs::toHoursBasis()` should, of course, be a product of the divisors, not the first divided by the second. Fixed code in johslarsen@6d9fa693a39f0ee6929b4c00de5e8f8f2aee788a

johslarsen, 2016-01-25:
Seems I never got around to squashing the previous commit, hence my changes with had me as author your original fix with you as author. `git blame` always shows author, `git log --pretty=full` shows both author and committer. `git commit --amend --author="Foo Bar <foobar@domain.tld>" to explicitly change author on previous commit (i.e. HEAD).

I have updates there the `#if` have been changed to only the else case:

* No history rewrite (i.e. new commit): johslarsen@d79ee798a2f95c2bd954e502b56aed71affb9e03 
* Me as author: johslarsen@3a2a16220ef772383b0ccaae8d9279e0b90b3064
* You as author, me as committer: johslarsen@994abf34bbd34c757f3f6a61b6183ea15f4e9b55

johslarsen, 2016-01-25:
I have tested the implementation at constant current discharge, and the energy accumulation calculation seems sane.

I_d = 1.0A
U(0mAh) = 3.64V (loaded)
U(1000mAh) @ 1.0A = 3.255V
E(1000mAh) = 3.43Wh = U_avg*I_d (given constant I_d)

If the discharge curve would have been linear (fairly good estimate for LiPo) then U_avg should have been 3.45, so pretty much spot on.

stawel, 2016-01-25:
nice, thanks for removing the tabulators :)

moved orgin/master to Me as author: johslarsen@3a2a162

> * You as author, me as committer: johslarsen@994abf34bbd34c757f3f6a61b6183ea15f4e9b55

hmm... that's interesting, I didn't know you can do that

johslarsen, 2016-01-25:
Yeah, my editors defaults to tabulators for C/C++ files, so they are only added accidentally.

Great, and I have gotten myself enough acquainted with this changeset that I have no problem being a possible scapegoat for it, despite you being the original author.

All in all I hate tampering with published history, but hey, this is your project, so you can have it your way. It should be mentioned that overwriting author is the exception, not the norm. The difference between committer and author originates from the era of distributing patches by mail, where the credit for a patch (i.e. author) is given to the original author and the intermediate (i.e. someone with write access to repository) that creates the actual commit, based on the emailed patch, is recorded as committer instead.

For reference here is the procedure I did to squash the commits:
  1. HEAD=energy_refactor_squashed (i.e. all my refactorings in a single commit after your original fix)
  2. `git rebase -i HEAD~2 # and mark the refactoring commit as fixup`, when completed you are author for the squashed commit
  3. `git commit --amend --author "Johannes Larsen <mail@johslarsen.net>"`, no changes, but updates previous commit with me as author

closed 2016-01-25 by johslarsen