stawel/cheali-charger #146: changeable battery names

stawel, 2016-01-27:


hryamzik, 2019-02-10:
@stawel do you have any ETA for this? Is that's what `Text=-` is supposed to display in battery status?

diggit, 2019-02-10:
AFAIK this feature is not possible on ATmega32 chargers due to not enough EEPROM space.
`Text` is reading of external temperature if you connect sensor.

specing, 2019-02-11:
There is only a couple dozen bytes left in eeprom. The easy way of doing this is by reducing MAX_PROGRAMS down from the current default, 20. E.g. MAX_PROGRAMS 10 frees up 400 bytes and would require 10*15 = 150 bytes for ascii description strings.

martonmiklos, 2019-08-14:
Hi @stawel 
I have used this feature on my M0517LBN based charger for a while, and now flashed a newer build an realized that this feature is no longer present. I saw you have removed this feature in 75176b0 . 

Would you accept a  PR which would selectively enable this feature if building for a charger with bigger EEPROM/dataflash?