stawel/cheali-charger #147: Accucel 6 dead 6A version

bisky00, 2016-02-02:
Hello,
Tried to flash my accucell 6, made backup of flash and eeprom, but not all the fuses. Charger after flashing does not work. Tried to flash back original firmware, display lightly blinks and buzzer buzzes continuously. Can anybody help?

Regards, Bisky

stawel, 2016-02-02:
> made backup of flash and eeprom ..

Your charger has probably a protected eeprom,
check if your backup contains some "meaningful" data (not only increasing numbers).

> Charger after flashing does not work

please see: https://github.com/stawel/cheali-charger#troubleshooting

bisky00, 2016-02-02:
This ar the numbers:
:2000000000000101020203030404050506060707080809090A0A0B0B0C0C0D0D0E0E0F0FF0
:2000200010101111121213131414151516161717181819191A1A1B1B1C1C1D1D1E1E1F1FD0
:2000400020202121222223232424252526262727282829292A2A2B2B2C2C2D2D2E2E2F2FB0
:2000600030303131323233333434353536363737383839393A3A3B3B3C3C3D3D3E3E3F3F90
So I gues the eprom was protected....

Will trie to update again,

TX.

bisky00, 2016-02-02:
Also the downloaded file did not heave hex content...

diggit, 2016-02-02:
- verify that mcu is responding `avrdude -p m32 -c usbasp`

If you get no errors, fuses are ok and MCU is capable of programming.

My Accucell 6 had locked flash. if you try to read it, you'll get just nonsense data.
To remove this lock, erase whole device: `avrdude -p m32 -c usbasp -e` (this is done before flashing if not explicitly disabled)

- get correct hex file for your charger from this repo and verify checksum, for now it is [hex](https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-Turnigy-Accucel-6-50W-5A_1.99-20160130_atmega32.hex) and [checksum](https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-Turnigy-Accucel-6-50W-5A_1.99-20160130_atmega32.hex.sha1)
- flash it: `avrdude -p m32 -c usbasp -U flash:w:your_file.hex`

in case your charger still doesn't work follow [Troubleshooting section](https://github.com/stawel/cheali-charger#troubleshooting) and set correct fuses.

If it does not solve your problem, maybe you have some different version of HW.

diggit, 2016-02-02:
if you have both files in same directory and with original name, use: `sha1sum -c your_file.sha1`, You shold get something like

`cheali-charger-Turnigy-Accucel-6-50W-5A_1.99-20160130_atmega32.hex: OK`

bisky00, 2016-02-02:
It seem to work now. Will do the calibration tomorrow.

TX, diggit and stawel

bisky00, 2016-02-03:
Calibrated, all the values were to high.

Thanks for the help guys!

closed 2016-02-03 by bisky00