stawel/cheali-charger #148: using Imax B6 as regulated power supply

bk79, 2016-02-04:
is it possible to add a feature so to use it as a regulated adjustable power supply? I'll explain better:
I set voltage and Max current and and It will feed power  accordingly

diggit, 2016-02-04:
Power part in these chargers is designed to work as constant current (CC) power supply. If you want to use your charger as CC, use battery type called LED.

Voltage regulation would be quite tricky, because you have to rely on ADC. Then MCU itself has to do the regulation. This would work for slow load changes. You can look into [powerSupply-test](https://github.com/stawel/cheali-charger/tree/powerSupply-test) and [powerSupply](https://github.com/stawel/cheali-charger/tree/powerSupply) branches.

This feature was discussed [[1]](https://groups.google.com/forum/#!searchin/cheali-charger/supply/cheali-charger/mPBPScJizLs/icrPG2SeEwAJ) [[2]](https://groups.google.com/forum/#!searchin/cheali-charger/supply/cheali-charger/8mwaktZM-00/mgS4coKNAQAJ) [[3]](https://groups.google.com/forum/#!searchin/cheali-charger/supply|sort:date/cheali-charger/ZpCBcNANEV4/73BHcL75CwAJ) [and more...](https://groups.google.com/forum/#!searchin/cheali-charger/supply)

bk79, 2016-02-05:
true unless you use a big capacitor at the output to keep the voltage up