stawel/cheali-charger #149: If cheali-charger support IMAX B6 mini?

3835, 2016-02-10:
Hi! 
First question is in the topic, and second question, how cooperation cheali-charger with LogView Studio? I ask because, now my charger "not talking" with this software, and i really think about new firmware:)

stawel, 2016-02-10:
1. no, we don't support Imax B6 mini yet
2. yes, we do support LogView (and LogView Studio)

3835, 2016-02-10:
#stawel thanks for quick answer and I keep my fingers crossed for cheali-charger development!

Milek7, 2016-03-12:
For dumping data from this charger, you can use my script: https://github.com/Milek7/imax-b6mini-datalogger/blob/master/imax-b6mini-datalogger.py
I don't know LogView, but if it supports .csv files it should work.

ireun, 2018-07-03:
@stawel  is it still not supported?

ireun, 2018-07-05:
Yup, You've explained that here: https://github.com/stawel/cheali-charger/issues/106#issuecomment-222520952 in ticket #106. Dziękuję mimo wszystko. ;)

stawel, 2018-07-05:
to ja dziękuję za commita :) i pozdrawiam

closed 2018-07-05 by stawel