stawel/cheali-charger #150: "Expanding" discharge support

stawel, 2016-02-25:
https://groups.google.com/forum/#!topic/cheali-charger/OBXQFJPIrpc

DerDominik, 2016-06-13:
Yesterday I'd the same question -> https://groups.google.com/forum/#!topic/cheali-charger/DUA4r62Bzlg
Is something planned to implement an extended discharging mode?

stawel, 2016-06-13:
> Sometimes I read about special charger where ~~you can connect external load (like a resistor, lamp or~~ maybe car battery (for recharge it)), if you need to discharge them.

you probably referring to:
> pg. 3 @ https://www.progressiverc.com/media/iCharger%20206B%20Manual.pdf

this is beyond the capabilities of cheap lipo chargers.

Regarding the "external load",
it should be possible, but as **diggit** said:
>Unfortunately this mode has some limitations. You need to have different load for each different cell count pack, then have to use it with balancer connected.

it's because we wouldn't able to control the discharge current
(It would only depend on the connected load),
additionally you charger should be able to measure the discharge current
(so there is a upper discharge current limit)
and some of the chargers have a limit of 300mA (but it's easy fixable/changeable).

>Is something planned to implement an extended discharging mode?

Unfortunately I don't have the time, to implement it in the near (probably even far) future,
but If you know C I highly encourage you to implement it yourself, It should be relatively simple to do,
You probably need to disable some safety routines and write a simple "discharge strategy".