stawel/cheali-charger #153: Generate images for documenation

burunduk, 2016-03-22:
Have refresh images to documents.
The archive contains the font and script to generate images
Script use ImageMagick for generate images from xpm fonts.

[xpm.zip](https://github.com/stawel/cheali-charger/files/184553/xpm.zip)
[image]

stawel, 2016-03-22:
wow!
can I add your script and fonts to the projects repository?

burunduk, 2016-03-22:
Yes, please.

stawel, 2016-03-22:
generator added to:
https://github.com/stawel/cheali-charger/tree/master/utils/lcdScreenshotGenerator

settings documentation looks much better now :)
https://github.com/stawel/cheali-charger/blob/master/docs/settings/settings.md

Thank you very much!

closed 2016-03-23 by burunduk