stawel/cheali-charger #160: Unable to start proocess

diggit, 2016-04-07:
It is not possible to start process when you start holding START button early, during transition between menu=>process specific screen.

Example:
- choose capacity check
- hit START btn (and release it)
- push and hold START btn as you want to start this process ASAP(yes we are impatient)
- process specific screen is displayed
- process does not start, I can list through all screens (ok, I'll release START btn and try holding it again)
- nothing happens, it does not start process, the only solution is to leave back to menu (STOP btn) and this time patiently waiting during menu->process spec. screen transition and then holding START btn to start process.

It's ok to ignore btn hold event during transition (MCU is probably busy), but I'd expect, that repeated pressing ang holding START btn will start the process.

stawel, 2016-04-11:
hm... interesting issue, 
it is however possible to start a program even it the scenario you are describing,
you just have to wait 0,7 seconds  with no button pressed.
But I get you point - unfortunately it's not easy fixable, or maybe it is and I need to think it over....

diggit, 2016-04-12:
Why that 0.7s? What is happening at this moment? I thought, that some kind of long press counter resets with pressing the button. (At least I do it this way in my transmitter FW.)

stawel, 2016-04-13:
in this particular case we check the buttons state in:
https://github.com/stawel/cheali-charger/blob/master/src/core/strategy/StartInfoStrategy.cpp#L94

The "Start Info" is just another charging "strategy" (that does nothing besides checking if battery is connected). Each "strategy" is called only when we collect all voltages and currents form all inputs - when we do a "full measurement", this happens every 0.7 second.

closed 2016-07-23 by stawel