stawel/cheali-charger #163: imax b6 as the power supply

13Mexicanec, 2016-04-25:
Good afternoon. I would like to know. you could implement in the future ( in firmware ) the possibility of using imax b6 (clone) as the power source to the stabilization and the ability to control the robots during both voltage and current . Thank you.

stawel, 2016-04-28:
>  I would like to know. you could implement in the future ( in firmware ) the possibility of using imax b6 (clone) as the power source

I'm not sure if it's possible at all,
If someone knows "control theory" I would be grateful to hear how to do it,
especially how to implement the voltage stabilization

FreifunkErzgebirge, 2016-11-24:
This is exactly what i was looking for.
BTW: Technically, it's possible and very easy with these chargers. I'm also at it currently. When i'm done, i will make a pull request.

diggit, 2017-01-09:
@FreifunkErzgebirge How is it going? Any success?

specing, 2017-06-09:
Just to clarify, does the above question mean using the Imax B6 as a general purpose DC-DC converter?

fiveangle, 2017-06-09:
I assumed it to mean using it as a general purpose, constant-current, constant-voltage, constant-wattage power supply.  This would be quite useful !

fiveangle, 2017-06-09:
if implemented, i think the project name would have to be changed to cheali-supercharger ;)

specing, 2017-06-09:
I found another issue here https://github.com/stawel/cheali-charger/issues/148, this time with a more understandable title.

reagentoo, 2017-11-14:
I charge my custom power bank with a built-in 4S balancing scheme. This BMS requires 18 volts to charge the batteries. I use the settings for 5S li-ion batteries to set 18 volts (without imax balancing). It would be very useful to have a separate setting for these purposes.
BMS:
https://ru.aliexpress.com/item/4S-8A-11-1V-li-ion-BMS-PCM-battery-protection-board-bms-pcm-with-balancing-for/32377617922.html
So far, I'm using the original firmware of Imax B6. And at the end of charging I see output at 28 volts. Apparently for some sort of calibration purposes. I think this is not very good for a balancing circuit.

fiveangle, 2017-11-15:
4S non-balance setting already exists

reagentoo, 2017-11-15:
> 4S non-balance setting already exists

4S 14.4v is not suitable with this BMS.

specing, 2017-11-16:
This problem is not relevant to the issue at hand.

However you can manually adjust the termination voltage for charging, iirc it goes up to 5V. The reason for it "requiring 18V"=4.5V/cell is that there is a drop due to the protection FET (and said FET must be awful if it drops 1.2V :D)

reagentoo, 2017-11-16:
@specing I received a response from seller "18V is maximum charge voltage, some people like more better balance effect.  in this case can use high voltage to charge the board. in short,  recommend is 4.2V*4= 16.8V".

But what about a calibration test after charging (up to 28 volts)? Is it possible to turn it off?

specing, 2017-11-16:
it might be avalanching, I know of no calibration test.

stawel, 2017-12-07:
>But what about a calibration test after charging (up to 28 volts)? Is it possible to turn it off?

we don't do any "calibration after charging", the 28volts you are seeing is probably due to
rapid change in output current  (occurs also when you disconnect battery wires during charging).
The charger is just a current source so it is unavoidable. You could try to change
 in file cheali-charger/hardware/targets/[your charger]/HardwareConfig.h :
#define MAX_CHARGE_V            ANALOG_VOLT(25.300)
to something smaller (but I don't thing it will help much, since this would mean we have a working powers supply).

qmr, 2020-01-20:
Clone charger EV-Peak C1 and friends have this feature, it's called "digital power".

Maybe decompile their firmware blob and take a look?

I would also like to see this feature.