stawel/cheali-charger #165: TVC Terminal Voltage Control

dapug, 2016-04-29:
I didn't see anything mentioned about TVC.  This is a top feature I am looking for, and the Accucel-6 80W actually has it, but can't run cheali-charger.   Not sure why the other chargers can't support this in firmware.  Thoughts?

Also, I'm new to cheali-charger, very intrigued to discover this, but not sure why there isnt a big statement on the **main page **that states **WHY** use cheali-charger versus the OEM firmware.   Adding features like TVC would be a big reason why.

stawel, 2016-04-29:
> I didn't see anything mentioned about TVC. This is a top feature I am looking for,

yes, we do have TVC (it's implemented in the "unstable" version, https://github.com/stawel/cheali-charger/tree/master/hex/unstable )

> Not sure why the other chargers can't support this in firmware. 

If your charger has a atmega32 or Nuvoton 0517 CPU then we probably support it,
(CPU type can differ within the same charger model)

> Also, I'm new to cheali-charger, very intrigued to discover this, but not sure why there isnt a big statement on the main page that states WHY use cheali-charger versus the OEM firmware.

yes, the readme is quite out of dated, I just don't have the time to update it.

dapug, 2016-04-29:
ok cool, nice work :)

closed 2016-04-29 by dapug