stawel/cheali-charger #166: GTPower A6-10 balancing error between port 1.99-20160419

PascalRZ, 2016-05-02:
Hi Pawel, hi all !

I have just uploaded my GTPower A6-10 from 1.00 to 1.99 version (cheali-charger-GTPowerA6-10_1.99-20160419_atmega32.hex)

I've a problem since 1.00 version : balancing function doesn't work.
If : 
- I connect a 3S LiPo with V2>V1>V3
- I start balancing program

I see : 
- bat1 flashing / bat2 flashing / bat3 not flashing
- V1 falling, V3 falling, V2 falling slowly.
- when V1 gets the lowest value, the charger continues to fall V1 and, maybe V3 (bat3 doesn't flash but V3 falls a lot)

I don't understand why, when V2 > V1 = V3, the charger doesn't fall V2 only. V1 balancing never stops and leads all values down.

What do you thing about that ? Software error ? Charger malfunction ?

A video (low quality) is attached in the zip file
[DSCF3476_1.zip](https://github.com/stawel/cheali-charger/files/245820/DSCF3476_1.zip)

PascalRZ

stawel, 2016-05-02:
interesting,
based on your video the software works as intended, namely:
1. determine cell X with lowest voltage (in your case cell 3)
2. discharge all other cells until all voltages go below voltage of cell X. (in your case cell 3)
additionally: we consider a voltage measurement as valid only when cells are not discharged
(icons are not flashing).

In your case, for some reason the cell 3 voltage is falling, so we try to discharge other cells to it's level
(and based on icon flashing we don't do anything to cell 3)

could you:
1. connect a voltage meter to the 3-rd cell and check if the voltage is really falling?
2. you could also measure if a current if flowing through the last balance wire 
   (2. -  may be hard to realize)

PascalRZ, 2016-05-04:
Good evening Pawel,

I made a special 3S balancing cable. Now, I can measure all voltages and currents independently. During my first use of the special cable, I used "balancing" but without the second validation (so measurement only, not balancing). I found a strange behavior (a video is attached):

When I disconnect wire 0 (gnd), the charger beeps and loses the voltage value of cell 1. OK
When I disconnect wire 1, the charger continue to measure all voltages, with a little difference between wire connected or opened. STRANGE ? I think it must lose cells 1 and 2.
Same thing for wire 2. STRANGE ? I think it must lose cells 2 and 3.
When I disconnect wire 3, the charger beeps and loses the voltage value of cell 3. OK

Do you think it could be a software issue, a fuse issue (I tried to troubleshoot with : " avrdude.exe -patmega32 -cusbasp -Uhfuse:w:0xc5:m -Ulfuse:w:0x3f:m " without change) or a short circuit in the charger ?

[DSCF3477.zip](https://github.com/stawel/cheali-charger/files/249311/DSCF3477.zip)

stawel, 2016-05-05:
> When I disconnect wire 0 (gnd), the charger beeps and loses the voltage value of cell 1. OK
When I disconnect wire 1, the charger continue to measure all voltages, with a little difference between wire connected or opened. STRANGE ? I think it must lose cells 1 and 2.
Same thing for wire 2. STRANGE ? I think it must lose cells 2 and 3.
When I disconnect wire 3, the charger beeps and loses the voltage value of cell 3. OK

This is behavior is actually normal: when you disconnect wire  1 or 2 the battery voltage is still applied
between wire 0 and wire 3. The charger has inside a balancing resistor+transistor ladder, which acts as a voltage divider when balancing is not running. This voltage divider will apply (more or less even spread) voltage  to wires 1,2 (even when nothing is connected to them).

see schematics:
https://drive.google.com/file/d/0B1RXXTatsA1cb1R5NHM3MEtsakE/edit

PascalRZ, 2016-05-05:
Hi,

I tried to read the schematics but I'm absolutely not an « expert » in electronic and I don't understand very well how this charger works (I will try more :) )

Using my old LiPo battery, cheali charger found during a “charge” R1 = 212 mOhm, R2 = 192 mOhm and R3 = 198 mOhm.

Using the balance program, but without starting the balance, I found :

wire 0 : -120 mA from batt to charger
wire 1 : -3,9 mA from BtoC
wire 2 : +3,5 mA from BtoC
wire 3 : +119 mA from BtoC

PS : my multimeter is a cheaper one, not very accurate.

Using the balance program, balancing started :

wire 0 : -120 mA BtoC, always
wire 1 : -4 mA when measuring, +27,8 mA if cell 1 picture flashes, and -8 mA if picture don't flash.
wire 2: +3,2 mA when measuring, +1,6mA if cell 2 picture flashes, not tested “if picture don't flash”.
Wire 3: I don't know because, when my multimeter is connected, the charger measures voltage but always beeps and refuses to start the balancing program. PS : when my multimeter is plugged on a wire, voltage measurement of the cell changes in cheali charger and go back to the old value when I remove the multimeter (effect of the internal resistance of the multimeter ?).

I tried another time with a new 3S pack. Currents changed just a little.

After reading your schematics, I found the 6 resistors, labeled “1300”, in parallel in each balance area and measured 23,1 Ohm for each “set”.
Also, I found “016H” near each set of resistors. It seems to be transistors but I can't find datasheet on the web (I will try more). I tested between the lower left pin and the middle right pin and found 600 mV for all the 6 “016H” using diode function.  My multimeter can't measure nothing from the upper left pin of each “016H”.

Do you need more tests ?

stawel, 2016-05-06:
>
> wire 0 : -120 mA from batt to charger
> wire 1 : -3,9 mA from BtoC
> wire 2 : +3,5 mA from BtoC
> wire 3 : +119 mA from BtoC
>

ok, this is wrong (assuming you didn't start the balancing program)
especially the ~120mA current.
could you do the measurement again but this time
with "main leads" wires disconnected?
(don't start the balancing program in all measurements)

I also need the voltages on attached picture,
where:
VBt1-VBt3 are measured between "arrow" and GND.
VCE1-VCE3 - as selected.

PascalRZ, 2016-05-08:
Pawel,
The new values are, in balancing mode but balancing not started :

With main lead connected :

wire 0 : -111,2 mA from battery to balance charger port
wire 1 : -3,8 mA
wire 2 : +2,7 mA
wire 3 : +112,0 mA

VCEt1 : -1,04 V
VCEt2 : -0,96 V
VCEt3 : -1,01 V

With main lead NOT connected :

wire 0 : -87,6 mA
wire 1 : -24,3 mA
wire 2 : +2,4 mA
wire 3 : +112 mA

VCEt1 : -1,67 V
VCEt2 : -0,96 V
VCEt3 : -1,01 V

I also checked the current from the main lead when it is connected : only 0,1 mA

I can't measure VBt1/2/3 because they are under the display and not accessible without creating a shortcut with my multimeter. If you really need these values, tell me yes and I will cut the 16 soldered display arms. I will have to wait 1 or 2 weeks for breakable pins (my stock is out).

I'm ready for others tests.

stawel, 2016-05-10:
hm.. I must admit that I'm quite confused

when balancing has not started yet all Darlington transistors should be turned off
(https://en.wikipedia.org/wiki/Darlington_transistor )

> VCEt1 : -1,04 V
VCEt2 : -0,96 V
VCEt3 : -1,01 V

so this voltages should be equal to Vb1, Vb2, Vb3 (voltages on balance port)
for some reason it's not the case (I assume your cell voltage is ~3V)
and that is why we see the current flow: 

> wire 0 : -111,2 mA
wire 1 : -3,8 mA
wire 2 : +2,7 mA
wire 3 : +112,0 mA

I don't understand why ALL off them are suddenly partly turned on.

Yes, the additional measurements will be nesesery, and I will need more data,
please measure:
1.  VBt1/2/3 
2. and the voltage between PB0 (pin 40) and GND (should be ~0V) 
3. PB1 (pin 41)  and GND  (should be ~0V)
4. PB2 (pin 42) and GND  (should be ~0V)
5. you could also check if GND of the amega32 is connected to the Q1, Q3,Q5 transistors GND

PascalRZ, 2016-05-10:
Ok,
I started to identify the different transistors. Now, I'm waiting for my electronic parts. I hope I'll receive the parcel in 2 weeks.
I will continue the job as soon as possible.

PascalRZ, 2016-06-04:
Good afternoon Pawel,
I've modified the charger and did the measurements:

WITH main leads:
Vbt1 = 2,82v
Vbt2 = 6,71v
Vbt3 = 10,49v
PB0 = 0v
PB1 = 0v
PB2 = 0v
Vcet1 = -1,03v
Vcet2 = -0,96v
Vcet3 = -1,01v

WITHOUT main leads:
Vbt1 = -0,85v
Vbt2 = 2,95v
Vbt3 = 6,70v
PB0 = 0v
PB1 = 0v
PB2 = 0v
Vcet1 = -1,64v
Vcet2 = -0,96v
Vcet3 = -1,01v

I confirm that GND of the amega32 is connected to the Q1, Q3,Q5 transistors GND.

I also confirm that, when I really start “balance” (long push), PB0 or PB1 or PB2 are at 0v when “m”, and independently are at 0v if cell picture is not fashing and are at 5v if cell picture is flashing.

Do you need an other test?

Best regards
Pascal

PascalRZ, 2016-06-05:
Hi,

In addition, I've a look on the Net about balance problem on the GTPower A6-10. I noticed that 2 versions have been manufactured : one just called « GTPower A6-10 » and the other one called also « GTPower A6-10 » with a sticker « New Kernel ».

Official website with only picture of the “first edition”:
http://www.gt-rc.com/en/pd.jsp?id=80&_pp=0_35&skeyword=A6

The pictures of the « first edition » PCB are already here :
https://github.com/stawel/cheali-charger/blob/master/docs/GTPowerA6-10/front.jpg
https://github.com/stawel/cheali-charger/blob/master/docs/GTPowerA6-10/back.jpg

Some explanations about balance accuracy adjustment can be found here:
https://www.hobbyking.com/hobbyking/forum/forum_posts.asp?TID=17200

Mine is a « New Kernel ». Two pictures are attached. Some parts of the PCB are different from the « first edition », and the balance area is different. Maybe the problem come from these differences?

[image]

[image]

I also found this video:
https://www.youtube.com/watch?v=NCBhwecispU
It's a “new kernel” and the owner of the video says the charger doesn't balance the cells with Cheali Charger.

stawel, 2016-06-06:
please make sure you measured the correct voltages, the fist cell is at the bottom (see pic)
[image]

I did the same measurements on my charger (https://github.com/stawel/cheali-charger/blob/master/docs/GTPowerA6-10/front.jpg <- this charger belongs to me, they shouldn't differ to much)
this is what I got:
I've connected a 3 cell battery to the balance port and main leads, voltages on cells:
Vb1 = 4.15V
Vb2 = 4.16V
Vb3 = 4.16V
when entering the "start info" screen in balance program (without starting the program) I got:
VCEt1 = 4.15V
VCEt2 = 4.16V
VCEt3 = 4.16V
(so the same voltage as on cells -> transistors are turned off, which is correct)
VBt1=4.07V
VBt2=8.15V
VBt3=12.23V
(voltages on transistors bases  are almost equal to cell voltages acording to GND,
 VBt1 ~= Vb1 = 4.15V
 VBt2 ~= Vb1+Vb2 = 8.31V 
 VBt3 ~= Vb1+Vb2+Vb3 = 12.47V
which is also correct).

Assuming your measurements are correct:
---------------------------------------------------------
for cell1:
1. The Darlington transistor is dead.
OR
2. Q1 - is dead (transistor controlling the Darlington transistor)
You could measure the Voltage on R3:
if it's ~=0V then the Darlington transistor is dead,
if VR3 >0V then Q1 is dead.
I'm not sure how big the voltage on R3 should be to sure,
so a better option would be to temporary desolder R3, if after that
VCEt1 ~= Vb1 (as in my case) the Q1 is dead otherwise the Darlington transistor.

You should do the same for cell2 to cell6.

stawel, 2016-06-06:
>Some explanations about balance accuracy adjustment can be found here:
https://www.hobbyking.com/hobbyking/forum/forum_posts.asp?TID=17200

hm... interesting, I even added such potentiometers to one of my chargers recently,
but the calibration method  described in this thread is not correct,
these pots are used to compensate for resistors tolerances used in the differential ap-amp circuit
(maximize CMRR http://www.analog.com/library/analogDialogue/archives/48-02/diff_amp.html)

> I also found this video:
https://www.youtube.com/watch?v=NCBhwecispU
It's a “new kernel” and the owner of the video says the charger doesn't balance the cells with Cheali Charger.

Yes, BalancePortAnalyzer can be used to check why a cell is not balanced,
it displays the cell voltages (ADC reading) and allows you to discharge
a particular cell (allows to turn on-off individual Darlington transistors),
but I'm quite skeptical whether it can help you in any way (your cells are discharged
even when the firmware turned off all Darlington transistors)

PascalRZ, 2016-06-06:
Pawel,
I started the investigations and found that, main and balance leads of my 3S battery connected, in balance mode but balance not started, Q11 is powered with 5v by the ATMEGA32 (output PA7 ADC7). Q1, Q3, Q5, Q7 and Q9 are not powered by the ATMEGA32.

Is it normal ?

stawel, 2016-06-06:
> Is it normal ?

it's definitely not normal,
are you sure atmeg32 pin 30 has 5V, all the time? 
("all the time" - I'm assuming nothing is connected to balance port input cell 6,
so you should never see a flashing cell 6 while balancing)

PascalRZ, 2016-06-07:
I controled pin 30 (PA7 ADC7): it stays always at 5v, as soon as I power on the charger and until I shut down it. I tried:
without nothing connected, in the main menu : +5v
with a 3S battery in "balance mode" ("start info" and "balance") : +5v
with a 3S battery in "charge mode" ("start info" and "charge") : +5v

stawel, 2016-06-07:
looks like your CPU is damaged, to be sure 
please download the newest firmware version (which I've  tested yesterday):
https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-GTPowerA6-10_1.99-20160607_atmega32.hex

You can also try to use the BalancePortAnalyzer-200W.

If the firmware update doesn't help you could desolder Q11
to check if the voltage on pin 30 is still 5V without the Q11 transistor,
if it is then you CPU is damaged and should be replaced.
(there is a very small chance that only pin 30 is damaged and
the rest of CPU still works, in this case you charger should work just fine 
with Q11 desoldered)

PascalRZ, 2016-06-11:
Pawel,

I tried the 1.99-20160607 version without success. I tried also the BalancePortAnalyzer-200W without success and finally, I tried a « stupid » version (voluntary charging the html instead of the real hex file) to check pin 30 with a not working program : pin 30 is dead, it stays always at 5v.

After that, I completely removed Q11, but all the balance ports stay working bad. I used the BalancePortAnalyzer-200W and found that all Darlington transistors work bad (not working at all or always activated). I removed (partially and completely) all the DT. I checked Q1, 3, 5, 7 and 9 and I think they work (voltage switches is “normal” when activated or not). R3+4, R13+14, R23+24, R33+34, R44+44, R53+54 are good too and I can't find short circuit in the balance area. I don't understand why DT seem to be all dead...

I found this documentation:
http://ru-radio-electr.livejournal.com/1295897.html

I don't speak Russian but I will try a translator.
If we use these scheme for identification number:
http://ic.pics.livejournal.com/electronik_irk/26265161/113638/113638_original.png
I found, with measure on VT22(Q11) completely removed and VT2,6,10,15,19 (Q1,3,5,7,9) soldered on the PCB, that the total internal resistance is 53kOhm and not 10k+10k as in https://cloud.githubusercontent.com/assets/199931/15072433/fd237216-1392-11e6-9816-15261cbdcda7.jpeg
so I think DTC114YKA (Rtot=10k+47k) is the good reference for these transistors in the GTPower A6-10.

I tried the charger without all the DT and Q11 and it works well (I charged a 3S battery).
I'm going to buy new DT, reference MMBTA64 and try to save balance ports 1 to 5 (6 is dead and I'm afraid of trying to change the Atmega32). It costs only 2 euros for 10 MMBTA64 replacement parts.

Thank you for your help. I never did smd electronic before. It's hard but interesting. I'll continue this issue after the new DT installation (delivery in 2 or 3 weeks).
During this period, I will try the 1.99-20160607 version and open new issue(s) if I found something (missing, improvement).

stawel, 2016-06-13:
>After that, I completely removed Q11, but all the balance ports stay working bad. I used the BalancePortAnalyzer-200W and found that all Darlington transistors work bad (not working at all or always activated)

You probably should fix this first, even if the pin 30 state should not affect other cells
(I've tested this yesterday on a working charger) in your case it seems
the atmega32 is damaged and this can affect other parts.
In my opinion you should check the pin 30 state again (measure the voltage on cpu) and
if you are really sure you have always 5V on this pin (especially with removed Q11),
you should replace the CPU.
The are some great youtube videos how to replace such a cpu with simple tools.

I also upgraded the hex/unstable/*BalancePortAnalyzer-200W* version today,
It had some problems with displaying voltages (ADC readings).

PascalRZ, 2016-07-15:
Hi Pawel,

I have received the MMBTA64 transistors. After soldering them and making direct solders between some resistances and the transistors (I suspect deterioration of some PCB wires), the balance port 1 to 5 works very well with GTPowerA6-10_2.00-e10.3.12-20160613. I’m very happy :)
The balance port n°6 doesn’t work because of the Atmega pin 30 failure. Now, I’m waiting for a new CPU but I’m scary about changing it. In fact, I don’t really need BalPort6 so I’m thinking about using the charger as this (for 1S to 5S batteries) or trying to change the Atmega with the risk of killing the charger…

In any case, thank you very much for your help.
Pascal