stawel/cheali-charger #168: original imax b6  discharge first cell inaccurate V1.99

GravityRZ, 2016-05-11:
Not sure if this is a bug.
i noticed in the manuals that during discharge of LiXx the first cell should read NA
mine reads a voltage but it is always 0.02 volts lower then the real value.
eg value is 4.200
display during discharge reads 4.180

the same goes for charging
in the first couple of minutes cell 1 read higher then cell 2 and 3 but during the charging cycles the are eqally balanced out.

also the voltage after charging seems ok.

could it be that cell 1 gets a higher load then the others during charge/discharge.

this could explain the drop in voltage during discharge and the rise in voltage during charge?

thanks

love v1.99

stawel, 2016-05-12:
> i noticed in the manuals that during **discharge** of LiXx the first cell should read NA
mine reads a voltage but it is always 0.03 volts lower then the real value.
eg value is 4.200
display during discharge reads 4.170

This is caused by a hardware design bug, you can little do about it.

>the same goes for **charging**
in the first couple of minutes cell 1 read higher then cell 2 and 3 but during the charging cycles the are eqally balanced out.

check if you have a yellow PTC element on your PCB near the balance port
(looks like a tall capacitor), if you do remove it.

> this could explain the drop in voltage during **discharge**

(assuming you have a imaxb6 charger)
during discharge the voltage on cell 1, more precisely on balance port negative wire (first wire)
drops below 0V, the ADC on atmega32 sees only voltages >= 0V, so this negative voltage is not visible, and:
Vb1 = Vb1_pin - Vb0_pin 
where:
Vb1 - cell 1 voltage
Vb1_pin - voltage on balance port second wire
Vb0_pin - voltage on balance port negative wire (first wire)

since Vb0_pin is negative, the firmware sees it as Vb0_pin = 0V 
that's why you see this error.

> and the rise in voltage during **charge**?

besides the usual measuring error on all cells you shouldn't see any additional errors.
if however you do have a PTC element  on your PCB,
then there is a current flow through the  balance ports negative wire (first wire) and the PTC element to GND, the current is  proportional to your main leads resistance (longer main wires - bigger current).
this current creates a voltage rise (V = current * balance port resistance) which adds to Vb1.

GravityRZ, 2016-05-12:
Hi Pawel,

Thanks fort he answers
I indeed have that PTC so i removed it.

voltage drop during discharge is still there(approx 0.3-0.4 volt)

however while charging cell 1 the correct voltage is displayed.

Why is that PTC there
probably to counteracts some temperature difference  but why is it needed  in the original firmware while with the cheali software it is counterproductive(creating false readings)

The software is really accurate.
The charger is balancing very accurate within 4 mv

Thanks

Ronald

stawel, 2016-05-13:
> voltage drop during discharge is still there(approx 0.3-0.4 volt)

yes, I is how it is, but  if you are using "adapt dis:" (adaptive discharge)
the discharge current will fall when you reach the end voltage
and the error is proportional to the discharge current, so it will also fall.

You could also just use the upper part of the balance connector
(the error shouldn't be visible on balance port cells 3 to 6).

> Why is that PTC there

I'm no sure why they add the PTC (no all charger have it, only the more expensive one).
It connects the first balance port wire (negative wire) with the minus main lead wire
on the chargers side.
The only reason why it may be useful is when you would like to balance your battery
without connecting the main leads (only balance port is connected), but we don't allow this anyway.

GravityRZ, 2016-05-13:
Thanks Pawel.
That is what i thought to.
it makes no sense since the original charger also would not let you charge without the main leads connected

closed 2016-05-13 by GravityRZ