stawel/cheali-charger #169: Turnigy Accucell-8 150W Li-xx ending issue (not 4.2V)

niarbagem, 2016-05-13:
Good day! 
Have Turnigy Accucell-8 150W (from hobbyking), had flashed to "cheali-charger-Turnigy-Accucel-8-150W-7A_1.00-20150711_atmega32.hex". 

Works well, but if i had charging Li-ion 3.7V (4.2V ends) with Li-Po 1S charging program, battery has only 4.17V after that. If had charged those battery on Opus BT-3100 v2.2 , and trying to watch it with Turnigy Accucell-8, it's read 4.22V.

Turnigy Accucell-8 was calibrated with Uni-T UT204 (had tested it with AD584kH Voltage Reference Module). 

May be i do something wrong?

stawel, 2016-05-13:
> May be i do something wrong?

you are doing everything right, 
cheali-charger just charges with a end voltage  of 4.20V
(maximum voltage applied to the battery wires is 4.20V 
  - we assume this approach is save for the battery)
That means it would take infinite time for a battery to be charged to exactly 4.20V.

We stop the charging process when the charge current goes down to "minIc:",
when we stop the battery voltage as to fall.
The simplest way to achieve a higher end voltage is to decrees "minIc:" to 50mA.
You could also increase the charging voltage but this is not recommended.

GravityRZ, 2016-05-13:
i noticed the same
i decreased min Ic to 100mA
i increased end voltage to 4.205
this way it settles exactly on 4.20 volts.

i also noticed different behaviour between multiple cell charge(balanced) and single cell charge.

i calibrated the charger with a 3S cell and charging, end result and displayed voltage is spot on(within 4mv)

however when i charge a single cell the end result on the cell is good(4.200) but the displayed voltage on the charger is lower.(4.190)
i corrected this by going to the expert menu and increased the vplus voltage with 0.1v so the displayed result and the voltage on the battery is the same.
this however seems to affect the voltage on cell 1 only when you do  a balancing charge afterwards

i

niarbagem, 2016-05-13:
Thanks, i'll see exper settings (where i can find it? i hadn't seen it at settings menu).

One more, first time i had calibrated charger with 2S Li-po (first 2 cells), second time -  with single Li-ion (with breadboard F-M cable for arduino) but all cells separately, so when i'd inserted 2s Li-Po  - second cell reading was wrong  (about 0.04V higher (if can remember)). 
And another - calibration question:
at calibrating process i had seen reading on charger had changed, when i touch the battery with multi-meter, so i must read the battery voltage on idle battery and correct it on charger or i must correct it on-the-fly (battery connected to charger and multi-meter)?

GravityRZ, 2016-05-13:
well i have the 1.99 firmware so hopefully you also

goto
options/calibrate/voltage
here you can calibrate your multicell lipo packs
it is best to use a regular 2s or 3s pack and not seperate cells like you did.
The reason for this is with a pack each cell has approx the same chemistry/capacity/resistance.
with 3 seperate cells you do not know what you get.

you can measure the voltages first and then connect later or you can connect the cells and measure the voltage from the balance pins inside(carefully do not short anything)
it does not make a difference. the cell voltage does not change when you connect the battery)
after this the balanced part is perfectly calibrated.

you can go back to the main menu(stop, stop, stop)

now hook up a single cell to the charge ports(18650, 26650 or some 1 cell lipo)
goto the programmed entry for that cell (eg lipo 1800mah/1
goto charge
now you can see what the voltage of the connected cell is
measure the voltage of the cell with your multimeter
you probably notice that they are not the same.
calcultate the difference in voltage(eg the charger is displaying 0.03 to high or to low)
goto the main menu (stop,stop,stop)
goto
options/calibrate/Expert settings danger.
now go down to the vplus voltage.
you see that this voltage is way of (something like 3.7 insead of 4.2)
now we do not calibrate these settings like advised
instead we just change them a bit
select vplus(it starts blinking)
now add or substract the voltage difference you measured earlier(0.03)
add/substract this voltage to the displayed voltage and press enter
go back to the main menu(stop,stop,stop)

this is another way of calibrating.
i noticed that following the normal procedure(hook up  a 1.5v battery) was not giving exact results.

so instead of calibtrating you just change the value a bit higher/lower and everything is perfect.

now my multicell packs are balanced equally
all voltages are correct(4.200) and correspond with the voltages displayed.

not bad for a 25mv asccurate charger to get it calibrated within 4mv

niarbagem, 2016-05-13:
Thanks, I'll try to correct by you method.

But I had seen voltage has changed not only for cell with voltage calibration but for Vin too, when I have touched cell or input with multimeter probe. I think probe wire resistance applied more load for atmega input?! Could it be?

And I think there is no matter calibrate with one cell or pack, main is to know accurate voltage for it, isn't it? But why I had seen difference for second cell with 2s pack and single cell? Maybe another unconnected pins makes difference?

GravityRZ, 2016-05-14:
Vin should not change once calibrated.
maybee your powersupply does not supply enough amps.
regarding the voltagre change on packs
with my charger(original imax b6) there is no difference between voltage on batterypack not connected or batterypack connected to charger (eg the charger does not affect the voltage)
maybee your charger does affect the battery voltage when connected.
if so you need to calibrate and measure when you connect the pack.

niarbagem, 2016-05-14:
Power supply is more than enough (Turnigy TGY 20Pro 220W) for my charger.
I hadn't  said  voltage on battery or Vin or battery pack was changed, but changed readings on charger during calibration process while i'd touched them (battery, Vin or battery pack cell) multimeter probes, Readins was different about 0,002V, examle: not touched - 3.834V (Li-Po 2S cell), touched (and hold) - 3.832V. 
 May you doesn't pay attention for this? I had noticed for this fact  only for third calibration process.

GravityRZ, 2016-05-15:
if you notice that then it must be your multimeter which affects the readings
0.002 is not bad. i notice that this is happening also when not touching the charger at all.

my charger has a resolution of 25mv and the fluctuation is 0.001 max which is good
what type of multimeter are you using?

i

niarbagem, 2016-05-17:
I have Uni-T UT204, it has 4000 count. But i had ordered Unit-T UT61E already which has 22000 counts for charger calibrating :) 
One more question i'd found: i have NiMH GP Batteries 20R8H 200mAh 8.4V , which needed to charge 20mA for 16 hours. On stock firmware minimum current was 100mA (it's too much for this battery), on cheali-charger i had charged it by 50mA (59 was reading), and battery was weak warm, it's great!
But charging has ended by capacity cutoff reason (240mA).For battery charging capacity was 200 edited. How capacity cutoff is calculating?

GravityRZ, 2016-05-17:
the capacity cutoff is a percentage setting in the charger
default it is set at 120%(which is 240) in your case.

i do not know if you can change it but try it

niarbagem, 2016-05-17:
I've seen that setting, but i want to know how it had calculated.
But this value is good. There was sometimes with stock firmware, charger was missed deltaV and battery (NiMH AAA size) was overcharged.

GravityRZ, 2016-05-17:
you better watch the Delta V because the lower the current the more chance of missing this delta.

i think default in cheali is 8mv but imax default is 7mv

closed 2017-01-14 by niarbagem