stawel/cheali-charger #171: hex and eep files

dankohs, 2016-05-22:
Please help. 
I can't find .hex or .eep files for eco8 charger. I am using eXtreme burner. How to download files needed to flash the charger?

Thanks

stawel, 2016-05-22:
if you have a atmega32 CPU then:
https://github.com/stawel/cheali-charger/blob/master/hex/unstable/cheali-charger-HobbyKing-ECO8-150W-7A-8S_1.99-20160419_atmega32.hex
(press the "RAW" button)

dankohs, 2016-05-23:
Thank you. But, i want to flash it so i can calibrate balance port on it, to unlock service menu1; i think that is called. For that i would need modified eeprom file? Chip is atmega32A.

stawel, 2016-05-24:
you mean the original firmware?
I don't know where you can find it, sorry,
but you should also ask on the cheali-charger@googlegroups.com mailing list.

closed 2016-07-23 by stawel