stawel/cheali-charger #172: How to get atmel processor

Mikimimi, 2016-06-02:
Can anyone tell is there any way how to buy imax b6 with atmel or other cheali-charger supported processors.
I'd like to buy one but don't want to get unsupported processor.
Thanks

alexFab, 2016-06-24:
recently i ordered two very cheap dc imax b6 clones from different sellers from ebay.

[image]

Both have an unlabeled processor. After some quick tests i have the strong feeling, the unlabeled processors are not atmels. Maybe there is a big batch of these clones without atmles on the Market right now.

Some month ago i ordered a IMAX B6AC 80W Dual Power AC that has an Atmel and works fine with Cheali-Charger

[image]

I am looking also for dc imax b6 charger with an atmel, maybe the safest way is to order the Original SkyRC imax b6

diggit, 2016-06-24:
There is no 100% rule how to get charger with supported MCU. Best chance is order charger from same seller as sb else. who got lucky and found one with supported MCU. But there is still chance, that seller has some mixed batch and you'll get unlabeled/unsupported MCU.