stawel/cheali-charger #173: [Semi-OT] Get Megawin MCU into bootloader mode?

nouser2013, 2016-06-04:
Greetings, perhaps someone can help me here. I wanted to update an accucell 6 80W with firmware 1.2 to 1.3, but something went wrong and it froze mit update. It comes with a built-in USB interface and has the MegaWin MG84FG516 MCU.

According to the [english datasheet](https://github.com/stawel/cheali-charger/files/298701/EN.MG84FG516_DataSheet_A1.5.pdf), there are three sections in Flash:

1. AP: regular flash
1. IAP: flash eeprom
1. ISP: code for bootloader

Since the ISP section cannot be overwritten by itself, it seems that this thing just does not enter the bootloader correctly anymore. Is there a chance to get it into bootloader mode without the use of the ISP flash tool (50 EUR, can buy 2 new chargers for that)?

Thanks

jief666, 2016-06-20:
Hi,
on a B6 mini (same CPU), holding "Left arrow" and "Start" and then plug the USB cable start in update mode (DFU). There is also a utility called DFU.exe in the development kit : http://www.megawin.com.tw/zh-cn/support/downloadDocument/117/316
Is there something like this on yours ? I'm interested to know if flashing via USB can debrick before I try a custom firmware that may brick mine. :-)
Jief

nouser2013, 2016-06-21:
Hi, thanks. According to the datasheet, there is a bootloader section which handles USB. Unfortunately, even though in theory not possible, this section got erased on my accucell. Therefore, pressing the key combo won't do any good. You'd really have to have one of their in-circuit programmers and those things are like 70 EUR + shipping. The whole charger costs way less.

> I'm interested to know if flashing via USB can debrick before I try a custom firmware that may brick mine

While I don't know how this could have happened (no power loss, and in theory, boot loader section is write protected from user app space), you apparently can brick it with no chance of recovery.

nouser2013, 2016-06-21:
Supplemental :) The bootloader code probably is still in there. But the only way to execute code from ISP flash section is through application code. If that got erased, the MCU will not enter ISP flash section and never execute the bootloader code. Crap by design ^^, so in case something goes wrong during flash, *afaik* that's it.

nouser2013, 2016-06-22:
Supplemental: I asked MegaWin about the issue and they responded with an undocumented way on putting the MCU into an HID USB mode in order to flash a firmware file ("DFU mode").

1. open case, top suffices
2. plug in usb cable (lcd on, mcu not doing anything)
3. see attached picture
  [image]
  * Short GND and OCD_SDA together and keep shorted
  * Short GND and RESET together, and release -> MCU boots into DFU mode
4. wait. Windows should recognize a new HID device and install standard drivers
5. load the new firmware from hobbyking website (for *Accucell 6 80W* http://www.hobbyking.com/hobbyking/store/uploads/1019411227X1816993X33.zip)
6. execute DFU.exe
7. load firmware file ("patched.bin")
8. press "upgrade"

after a while, the progress should be indicated as text in the text box bottom left.

hope that this helps someone some time in the future...

jief666, 2016-06-22:
Hey,

amazing ! Did you do it ? Do that works like they say ?

I'd like modify and compile a firmware and try on mine. But for sure, won't work first time => crash and need hardware way of reflash.

Do you know that there is MOSI and MISO pins (pin 2 & 3) ? That the way they flash atmega with a cheap device. Anyone have an occasion to try ?

Jief

nouser2013, 2016-06-22:
> amazing ! Did you do it ? Do that works like they say ?

Yes, as you can see in the picture, the charger is operational again.

> Do you know that there is MOSI and MISO pins (pin 2 & 3) ? That the way they flash atmega with a cheap device. Anyone have an occasion to try?

Wait. My description is for a MegaWin MCU. If you have a charger with AVR MCU (completely different manufacturer) you don't have a problem at all, you can just buy a cheap AVR in system programmer off eBay.

And regarding the Accucell 6, there are other versions with an AVR MCU design. But they have the 6 pin AVR ISP pin out. See this image:
[image]

The difficulty here was that the MegaWin ISP tool is impossible to come by in my country.

jief666, 2016-06-22:
No, I have a MegaWin exactly like yours. But there is the same pins MOSI & MISO as on Atmega. So maybe it can also be programmed with the same device.
One way is enough. If this one works, I may start to brick mine. :-)

nouser2013, 2016-06-23:
> But there is the same pins MOSI & MISO as on Atmega. So maybe it can also be programmed with the same device.

I don't think so. Their ISP uses !RESET, OCD_SCL and OCD_SDA signals. I'm fairly certain that they did not build the same ISP protocol as Atmel into their devices, moreover since this is intellectual property of Atmel (or MicroChip now).

jief666, 2016-06-23:
I've download the datasheet from their website : http://www.megawin.com.tw/zh-cn/support/downloadDocument/119/381
and there is that is it :
[image]
Just for information, if you are in contact with MegaWin, ask what are they for ?

diggit, 2016-06-23:
1) MISO and MOSI are common names for SPI (Serial Peripherial Interface), they are not dedicated programming pins. It's just that AVR ISP programming uses SPI to communicate with programmer.
2) there is not support for 8051 architecture in cheali charger -> you have to stick with stock fw, or port it by yourself. [Look here](https://github.com/stawel/cheali-charger/issues/106)

jief666, 2016-06-23:
1) Thanks for information.
2) I know. My plan is to develop (that's why I'm sure to crash it sometime). I know there is a branch that compiles with SDCC. I'll start with that.
I'd like to do a small firmware that converts this into a Constant-Voltage Constant-Current small power supply, for example.

lostcontrol, 2016-10-28:
I bricked a SkyRC iMax B6 mini a year ago while upgrading the firmware. Wonder if I can resurrect it if I manage to put it in bootloader mode!? That small charger was pretty handy...

jief666, 2016-10-28:
Let me know, I'm interested !!

lostcontrol, 2016-10-28:
Didn't work :-(

diggit, 2016-10-28:
Were you able to put it into bootloader mode? What failed?

lostcontrol, 2016-10-28:
No, nothing happened. No HID device detected by my Linux laptop. When I power on the charger, the first row of the LCD is filled by black squares, the second row is empty. Nothing more, nothing less. Don't know how badly it is bricked. The PCB is not the same as on the picture above but the pins look exactly the same. Any idea?

jief666, 2016-10-28:
Maybe not the same order. The reset might not be the middle one ?

nouser2013, 2016-10-28:
You could try the other pin, I may have mixed the two OCD_* descriptions. But still:

1. Power via USB (LCD stuck, not doing anything)
2. Connect GND and OCD_SDL/SDA (one of those, but not both at the same time); best solder a short wire
3. Pull !Reset line by connecting it shortly to GND
4. USB enumeration happens almost instantly

Again: does *not* work for AVR MCUs, only for MegaWin MCU.

lostcontrol, 2016-11-02:
No, nothing. That damn thing doesn't react to anything... I tried many combinations (even GND/+5V by mistake) but nothing.

nouser2013, 2016-11-02:
well, to quote a very popular TV show character:

> It's dead, Jim.

Maik2016, 2016-11-16:
You have to pull reset short to +5v  then it works.

After this you can use every Firmware tool (DFU)

aheyer, 2017-06-26:
I can confirm: Shortly connect !Reset (inverted!) to +5V while holding GND to OCD_SCA to enter USB bootloader. I had to do it on a Turnigy Accucell 6 50W 6A **LiHV**, that does not work with the provided firmware for the older Atmel Non-LiHV version. The ISP pins of the LiHV version are at the bottom side of the board, not at top the LCD side. Unscrew the board and you'll find 5 solder holes in a row with two marked as + (5V) and - (GND). !Reset is next to + and OCD_SCA is the last hole in the row when starting with GND from left to right.

LTMNO, 2018-09-05:
I can share that i followed this process... on my MegaWin CPU... i was able to get into DFU mode.
I updated the firmware via the USB with B6AC_Upgrade_V1.14_Add_Calibration Firmware.

S474N, 2023-06-25:
> You have to pull reset short to +5v then it works.
> 
> After this you can use every Firmware tool (DFU)

Same problem, and also with +5V works!

[image]

After that flashed to newest 1.17:
[image]