stawel/cheali-charger #174: IMAX B6 clone: "connection break" at charge

mkdima, 2016-06-09:
Sorry for my ignorance, I know that it's a different firmware (the default one on iMAX clone) but I can't seem to find the HW fault. I was wondering, can you point me to the right source files to look at? What would be the conditions that are monitored and would produce that error? I know what it means, but I guess that the firmware monitors certain signals before throwing that error. If I would know exactly what, I can do some more HW debugging. At this point I'm kind of stuck. Any help is greatly appreciated. Thanks

stawel, 2016-06-09:
I'm not familiar with stock firmware error messages,
the only thing I can do is point you cheali-charger "monitor" routines,
but I doubt that they have something in common.

monitor routine:
https://github.com/stawel/cheali-charger/blob/master/src/core/strategy/Monitor.cpp#L208-L279

displayed strings:
https://github.com/stawel/cheali-charger/blob/master/src/core/strings/standard.h#L43-L53

If you would flash you charger with cheali-charger firmware,
I could probably help you more.

mkdima, 2016-06-09:
Thank you very much stawel, I really appreciate your quick response. I guess I'll do this next (update the firmware). I got it off ebay, seller in China, came broken, can't seem to convince the seller for  a refund or replace. It was completely dead when I got it. In the meantime I found out that voltage reg. L7805 was broken. After I replaced it it's alive but throws that error whenever I try to charge any battery, and I'm not able to find any more faults. Thanks again!!!

stawel, 2016-06-09:
>Thank you very much stawel, I really appreciate your quick response.

it's just a coincident, unfortunately most of the time my response takes some time ;)

> I found out that voltage reg. L7805 was broken

it's probably the worst possible case, the L7805 could damaged many more parts.

> I guess I'll do this next (update the firmware)

after that check if your voltages are correctly measured:
1. go to: "options"->"calibration"->"voltage"
2. connect a battery to main leads and balance port 
3. check is Vin is approximately equal to you power supply voltage
4. press the "down" button until you see:
      V1-6:     12.420V
      Vout:     11.871V
5. check is the two values above are approximately equal (they can differ a couple of volts)
6. do not calibrate voltages yet (don't press the "start" button in this menu)

mkdima, 2016-06-11:
Thanks very much. I uploaded "cheali-charger-imaxB6-original_1.00-20150711_atmega32" on it.Is it OK to use this firmware, although it's described being 200W on your download page, but my charger is only 50W,? Otherwise I followed your steps, and it seems that the problem is in the balance port (I'm using a 3cell lipo) and it shows Vb1=0, Vb2=7.24, Vb3=3.9, Vb4=0.06, vb5=0.034, vb6=0.021 and V1-6=0. The Vin and Vout show correct values (12.8V, 11.5V).
I guess that means there's a disconnect somewhere on the balancing sampling path.

diggit, 2016-06-11:
@mkdima use most recent and proper version from [hex/unstable/](https://github.com/stawel/cheali-charger/blob/master/hex/unstable/), currently [this file](https://raw.githubusercontent.com/stawel/cheali-charger/master/hex/unstable/cheali-charger-imaxB6-clone_1.99-20160607_atmega32.hex).

High power chargers have slightly different cell measuring circuit and their builds are not compatible with low power ones.

mkdima, 2016-06-13:
I did a BAD thing.. for some reason while power was off but the balancing port was connected to the battery, and I was measuring the voltages on those ports, I might have done a short or something but suddenly the LCD turned on, and off. When I then powered it back on the LCD shows white squares and Atmega is not responding anymore. I guess that somehow the 11V from BATT went over the 5V rail and burned the micro-controller. I was puzzled, even now I don't understand how that happened. So now I'm waiting for a new Atmega, will try to solder it back, reprogram etc..lots of "fun". Hopefully I saved the eeprom/flash/fuse, do you think that those bin files are enough to make a blank Atmega to work, or there's more to it? Thanks..

stawel, 2016-06-13:
>lots of "fun"

;)

> do you think that those bin files are enough to make a blank Atmega to work, or there's more to it? 

you don't even need these bin files, everything is freely available

closed 2016-07-23 by stawel