stawel/cheali-charger #175: Idea of improvement : completion ring setting

PascalRZ, 2016-06-11:
Hi Pawel, hi all,

This is an idea of improvement:

On my Icharger 106B+, in the general setting, there is an option called “Completion Ring” with 4 choices:

Beep 5 times
Beep 3 minutes
Beep always
Beep off

Actually, cheali charger beep only 1 time at the end of charge, so if you are not always near the charger, you can't know if the charge is completed without having a look at the screen. With the option “always” you can listed continuously the end of charge and you don't have to go back to look at the screen.

Adding “Completion Ring” in general setting with options “beep 1 time” or “beep always” will be a good improvement.

In option (because it's easy to ask for something but I don't know how to improve the program by myself):
“Beep off” (stupid?)
“Beep 3 minutes” (because if you are far away and you use “beep always”, people around the charger could become nervous)

An other idea: turning backlight screen on and off each second at the end of charge.

Pushing one of the 4 buttons stop beeping and flashing.

What do you think about this idea?
Pascal

stawel, 2016-06-13:
I like your proposal, especially:
>An other idea: turning backlight screen on and off each second at the end of charge.

but due to lack of enough spare time, you proposal will have to wait for a volunteer :(

mkdima, 2016-06-14:
Turning backlight off seems a good idea but it may not be possible on all chargers. Eg. on IMAX B6 the backlight is always connected to 5V (either through a transistor for the original, or directly for the clones).