stawel/cheali-charger #182: add to readme

stawel, 2016-06-24:
https://github.com/stawel/cheali-charger/issues/172#issuecomment-228463427

diggit, 2016-06-25:
I suggest adding line from troubleshooting section "download the \*.hex again, use the "RAW" button in github" somewhere higher. There are many cases, when people does not use RAW button, but downloads directly link from directory listing.

Maybe some section like **getting ready** /guidepost/TOC
- get compatible charger (and add link to HW section..)
- get programmer (and simple description or link)
- get multimeter (1mV resolution around 4V scale recommended)
- setup SW (linux, windows, mac? - just avrdude or some gui frontend)
- download correct flash image and how to do it (RAW btn.) 
- flash (link to flashing section)
- calibration (link to calibration section)

Yeah, just TOC will help a lot.

What do you think?
Then, I can do changes and open PR if you want.

stawel, 2016-07-22:
Yes, I need to rewrite the whole readme :( - unfortunately don't have time for that.
but If you want to change it yourself I'll definitely accept any changes when it comes to the readme :)