stawel/cheali-charger #183: Turnigy B6 PRO 50W 6A - supported?

ndandanov, 2016-06-29:
Dear All,

Congratulations for the great initiative on this alternative firmware!
Has anybody tried to flash the cheali-charger to a Turnigy B6 PRO 50W 6A charger?

It is available at HobbyKing:
[[image]](http://www.hobbyking.com/hobbyking/store/__58319__Turnigy_B6_PRO_50W_6A_Balance_Charger_Blue_.html)

According to this discussion, it should be supported:
https://groups.google.com/forum/#!topic/cheali-charger/hfMrRio8c9U

Some detailed pictures can be found here:
Back: [[image]](https://m.flickr.com/#/photos/34067555@N00/16365954471/)
Front: 
[[image]](https://m.flickr.com/#/photos/34067555@N00/16181471159/)
CPU: 
[[image]](https://m.flickr.com/#/photos/34067555@N00/16366774462/)

The CPU seems to be an Atmega32.

Thank you in advance!

Kind regards,
Nikolay

stawel, 2016-07-23:
Have you tried cheali-charger on it?

fiveangle, 2017-04-04:
I got it to work (those are my pictures reposted above from the linked thread :) but it only ran at 5A, not the supported 6A

-=dave

diggit, 2017-04-04:
@fiveangle limits are compiled in FW. You can select different FW image with appropriate limits or modify sources and build it by yourself.

fiveangle, 2017-04-04:
I gave the charger to a friend as I found it got too hot for my liking, even at 5A :)

greenbigfrog, 2017-04-09:
(is there a reason why this is still open?)

diggit, 2017-04-09:
@stawel is quite busy person. I believe, he will close this issue as soon as he finds spare time to do so.

fiveangle, 2017-04-14:
@ndandanov could close it...

stefan-ro, 2017-05-10:
I bought my Turnigy B6 Pro two weeks ago in the EU Warehouse. And I received a version with a nuvoton CPU. But with a new board layout i haven't seen yet. 

The board has a new pin-layout an the nuvoton CPU has rotated. Beside this the crystal is 12MHz and not 16MHz.

[image]

I already identified the programming header (at the bottom of the board)... but I need help setting the new pin layout in the code.

greetings
stefan

diggit, 2017-05-10:
Hi @stefan-ro, look into #207. I believe. you have same HW.

stefan-ro, 2017-05-10:
Hi @diggit, thanks for the reply... the HW seems to be very similar like in [https://github.com/stawel/cheali-charger/issues/207](url). But I have another HW Revision... and comparing the photos I found some (minor) differences. 

[image]

I will checkout Ralim's fork and setup an arm toolchain on my machine....

ndandanov, 2025-03-04:
`Turnigy Accucel-6 50W 5A Balancer/Charger w/ Accessories` is written as supported in the [Hardware section](https://github.com/stawel/cheali-charger?tab=readme-ov-file#hardware), if it has an Atmega32 CPU.

I also think this issue is quite old and can be thus closed. :-)