stawel/cheali-charger #185: Support Titan batteries (charge cutoff 4.2V, discharge cutoff 2.5V)

Googulator, 2016-07-18:
The Titan battery (https://titan-flight.myshopify.com/collections/endurance/products/3-5ah-14-8v-80w-endurance-brick-version) is supposed to be charged to 4.2V and discharged to 2.5V per cell. Other NCA-chemistry batteries almost always have the same limits.

Cheali doesn't currently offer a mode for this. LiIo discharges to the correct 2.5V, but only charges to 4.1V, while LiPo has the correct charge limit, but wrong discharge limit.

diggit, 2016-07-18:
You can set these thresholds manually. You just need to enable advanced menus a then *edit battery*. I don't know if there is enough space to accommodate another cell type template.

diggit, 2017-07-04:
If manual limit settings is ok for you, can you close this issue?

fiveangle, 2017-07-04:
> I don't know if there is enough space to accommodate another cell type template.

you bring up a good point - LiLo, LiFe, LiPo, LiPoHV, these are all identical charge charictersitcs (CC->CV) differing only in their start/end voltage limits.  I would imagine they are stored similarly ?  I'd imagine just a dozen bytes or so per type ?

I was trying to find places to reduce mem usage, to make room for a "simplified" interface at the top-level menu, which emulates a limited-functionality 4-button interface we're used to, for when needing to do utility charging of random batteries.

diggit, 2017-07-04:
> I'd imagine just a dozen bytes or so per type ?

Too lazy to search in code, but  I guess sth like 22 bytes per type or so. (in flash)

> I was trying to find places to reduce mem usage, to make room for a "simplified" interface at the top-level menu, which emulates a limited-functionality 4-button interface we're used to, for when needing to do utility charging of random batteries.

Interesting idea. Not sure what to shrink or omit to save flash memory. I'd be nice to see more ARM based chargers with sufficient memory and ADC for cheali.