stawel/cheali-charger #188: time limit in code?

tsaitgaist, 2016-07-21:
I am using an iMaxB6 clone based on a nuvoton-M0517 using firmware cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_nuvoton-M0517.hex for testing the capacity of a 12 cells 2200mAh Ni-MH battery.
The capacity test somehow fails. When I come back after some time I don't see the capacity results. Instead the after power on screen is displayed. I can reproduce the issue every time. The time limit is set to "nolimit". 
I then used the logview output to see if something suspicious (e.g. too high temperature) occurs. I found that the crash happens around 31k seconds (not always the same time).
Here the last line: $1;9;31554.3;18504;1883;2122;3484;3646;61647;4591;18$&^D^@^M
You can see the crash since the line isn't complete.
~31k sound to me very suspiciously like a signed int16 integer overflow issue. I don't know the code and could not pinpoint though.

stawel, 2016-07-22:
hm.. 8 hours o charging...
ok I will try to reproduce and fix your issue, but I will  take some time :)

If you do have the full logs, preferable plain text, I would like to see them.

What exactly do you mean by: "power on screen" is it:  "cheali-charger|v2.00-20160613"?
and does the charger respond to button presses?

tsaitgaist, 2016-07-23:
when it crashes it restarts normally. it shows cheali-charger|v2.00-20160613 and goes to the menu. everything (including buttons) work as before and I can restart the capacity test.
[Uploading imaxb6-crash.txt…](Here) the complete log.

stawel, 2016-07-23:
hm.. I'm currently at:
$1;1;33840.6;2969;76;714;22;203;48073;3453;14035;0;0;6;5;6;6;0;0;0;0;0;0;3600;0;60;30824;58
and charger is still working,

do You have more or of these logs? If yes, I would like to see them too,
and what power supply are you using?

tsaitgaist, 2016-07-24:
I am using a HP PP018H 19V 1.58A laptop power supply.
I don't have the other capacity logs anymore, but I just started a new one. I'll upload it when complete.
Meanwhile I've also tested the D>C format action. The charger did not crash, and there is no ~30k integer overflow issue. But there is a uint16 wraparound issue. Here the 
[log](https://github.com/stawel/cheali-charger/files/380219/imaxb6.log.zip).

stawel, 2016-07-24:
> But there is a uint16 wraparound issue. Here the 

indeed, bug fixed in develop branch

stawel, 2016-07-24:
> I am using a HP PP018H 19V 1.58A laptop power supply.

I have a guess what is happening, your charging plot:
[image]

magnified:
[image]

There is a small power supply voltage drop,
You are charging with 37W, the charger is consuming even more and your power supply is rated at 30W, It is probably shutting down after a while

tsaitgaist, 2016-07-25:
You are right. I should have looked for it. Sorry for the false alarm.
I've reproduced the error and the power supply dip is there. Curiously the power supply does not shut down immediately when > 30W is drown. You can see it during the first charge (does not shut down), and the second (shuts down after some time). With a 19V 4.22A power supply I have no issue. Here the [logs](https://github.com/stawel/cheali-charger/files/381003/power-supply.zip).

A nice feature to add to the firmware would be to detect this fault (sudden drop of 0.2V in Vin when charging), stop, warn the user about the power supply not delivering enough current, and suggest to lower the charge current setting (for this battery). If the power supply still shuts down (powering off the device), the iMaxB6 could save the error in flash when detected, and display it at boot, until the user confirms the error.
But that is just a feature idea. Thanks for all the good work.