stawel/cheali-charger #190: Autosstart possible?

UserChristian, 2016-07-22:
I want to use the charger on a robot mower project.
The charger self is integrated into the mower and if the batterie is empty the mower is driving back to the base station. Then it would be nice to start the charging process direct after connecting to the power source with the last selected batterie or one stored batterie settings.
Is it possible to integrate that feature?

stawel, 2016-07-22:
I would recommend you to implement it yourself, since I running out of spear time.
see:
https://groups.google.com/forum/#!msg/cheali-charger/D3_OvkyY4z8/67key5frbOcJ

If you have any questions just ask - unfortunately it  takes me some time to answer.
also if you have some pics for your "robot mower project" it would be nice to see them :)

zpukr, 2023-10-16:
I implemented a fork with auto start support, source files and ready-flash hex firmware on GitHub