stawel/cheali-charger #196: Turnigy Mega 400W X2 I calibration problem

daerias, 2016-09-06:
Hi :) 
It works great till calibration of Ic & Id. After that I have calibration error 11 or 13. Please help. Greets Darius

stawel, 2016-09-06:
1. reset your calibration  "options"->"reset default" (this will reset all your settings) 
2. try to calibrate only Ic 
3. check if charger works
4. try to calibrate  Id 
5. check if charger works
6. calibrate voltage
7. check if charger works
8. if you still get an error see: https://github.com/stawel/cheali-charger/blob/master/docs/calibration_error_codes.md
(although error 13 means your are doing something wrong)

daerias, 2026-08-30:
Closing this — it's from 2016 and no longer relevant for me. Thanks for a great project.

closed 2026-08-30 by daerias