stawel/cheali-charger #197: Accucell 8150 bad calibration on Vplus pin ?

Tomin123, 2016-10-01:
Hi,
I try cheali on my 8150 charger and have some trouble I don't understand.
I did basic calibrations (on 2cells Lipol) and setting for Vout seems exactly the same like V1-8
Current calib. seems nice too.

After that I tried charging 2lipol and got error:
calibration error
Voltage      4

Tried reset of settings and start from begin twiced, no help.

So I started read sources and realize that reason "Voltage 4" means something with Vplus calibration.
I am right ??
So I entered into an expert menu of calibration and tried calibrate Vplus.
Once I click on Vplus option (there was some value around 6.xxx Volt I think)
it shows value of 36.300V and staid there. I can not change it at now even with long time hold any increace/decrease button.
(I checked voltage on Pin 37 of AVR (ADC0, Bat+) and there is 0.58Volt.)

What cause could be there, please ?

Thank you very much in advance.
Tomas

BTW: thanks for your very nice work

stawel, 2016-10-01:
what was the voltage on the battery when:
> (I checked voltage on Pin 37 of AVR (ADC0, Bat+) and there is 0.58Volt.)

stawel, 2016-10-01:
Unfortunately you will have to recompile the source,
you need to change:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/Turnigy-Accucel-8-150W-7A/HardwareConfig.h#L34
to something smaller, try:
#define MAX_CHARGE_V            ANALOG_VOLT(35.000)
(although 2.5V/0.58V*8.4V ~= 36.2V should probably work)

after compilation you should see a new hex file in src/hardware/atmega32/targets/Turnigy-Accucel-8-150W-7A/

Tomin123, 2016-10-02:
Thanks for help. I will try it.
There was 7.4Volt on the battery.
(I can modify simply a voltage divider on an input, if it could help. Should I ?)

Tomin123, 2016-10-02:
I tried recompile, but CMAKE is over my head :-)
Install it, put it into path etc.
(I am on WinXP)
It seems like all went ok, but nothing is recompiled :-(
How can I direct cmake to strong recompile all ?
```

C:\temp\cheali>set CC=c:\winavr\bin 

C:\temp\cheali>cmake . -DCMAKE_TOOLCHAIN_FILE=avr-toolchain.cmake -G "Unix Makefiles" 
-- Could NOT find UnixCommands (missing:  BASH) 
-- setting up cheali-charger MACROS.
-- cheali-charger-version: 2.00
-- cheali-charger-eeprom-calibration-version: 10
-- cheali-charger-eeprom-programdata-version: 3
-- cheali-charger-eeprom-settings-version: 12
-- cheali-charger-eeprom-version-string: e10.3.12
-- cheali-charger-buildnumber: 20161002
-- target architecture: avr
-- target: ADCKeyboardAnalyzer-150W
-- cpu: atmega32
-- generic charger: adc_key_test
-- target: BalancePortAnalyzer-200W
-- cpu: atmega32
-- generic charger: 200W
-- target: Dual-Power-B6AC-80W-RC
-- cpu: atmega32
-- generic charger: 50W
-- target: imaxB6-clone
-- cpu: atmega32
-- generic charger: 50W
-- target: imaxB6-clone-50W-4.5A
-- cpu: atmega32
-- generic charger: 50W
-- target: Turnigy-A-6-10-200W
-- cpu: atmega32
-- generic charger: 200W
-- target: Bantam-BC6HP-250W
-- cpu: atmega32
-- generic charger: 200W
-- target: Turnigy-Accucel-8-150W-7A
-- cpu: atmega32
-- generic charger: 200W
-- target: GTPowerA6-10
-- cpu: atmega32
-- generic charger: 200W
-- target: imaxB6-original
-- cpu: atmega32
-- generic charger: 50W
-- target: Turnigy-Accucel-6-50W-5A
-- cpu: atmega32
-- generic charger: 50W
-- target: Turnigy-MEGA-400Wx2
-- cpu: atmega32
-- generic charger: 200W
-- target: Turnigy-MEGA-200Wx2
-- cpu: atmega32
-- generic charger: 200W
-- target: Turnigy-MAX80W-7A
-- cpu: atmega32
-- generic charger: 50W
-- target: Turnigy-Mega-300W-8s
-- cpu: atmega32
-- generic charger: 200W
-- target: HobbyKing-ECO8-150W-7A-8S
-- cpu: atmega32
-- generic charger: 200W
-- Configuring done
-- Generating done
-- Build files have been written to: C:/temp/cheali

C:\temp\cheali>
```

stawel, 2016-10-02:
> C:/temp/cheali/src/hardware/atmega32/cpu/memory.h:158: error: there are no argum
ents to 'eeprom_update_word' that depend on a template parameter, so a declarati
on of 'eeprom_update_word' must be available

the 'eeprom_update_word' was not declared:
http://www.nongnu.org/avr-libc/user-manual/group__avr__eeprom.html
it should be declared at #include \<avr/eeprom.h\> (line 22)
try to find where your winavr installed the file avr/eeprom.h
and add its path to avr-compiler.cmake, change line 11:
SET(CINCS "")
to 
SET(CINCS "-Ic:/winavr/some/path/")

Tomin123, 2016-10-02:
Many thanks for your fast support !
Done. I had older version of WinAVR (2007 ??), sorry.
So there were not *_update functions at all.
Reinstal WinAVR to 20100110 (hope latest) and it started compiling.
And after some work I got another error :-(

part of log taken from screen:
```
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:224: warning: only initialized
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:259: warning: only initialized
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:272: warning: only initialized
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:464: warning: only initialized
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:471: warning: only initialized
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:478: warning: only initialized
 variables can be placed into program memory area
C:/temp/cheali/src/core/calibration/Calibrate.cpp:479: warning: only initialized
 variables can be placed into program memory area
[  3%] Building CXX object src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150
W/CMakeFiles/cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atme
ga32.dir/__/__/__/__/core/menus/StaticEditMenu.cpp.obj
[  3%] Building CXX object src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150
W/CMakeFiles/cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atme
ga32.dir/__/__/__/__/core/menus/EditMenu.cpp.obj
[  3%] Building CXX object src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150
W/CMakeFiles/cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atme
ga32.dir/__/__/__/__/core/menus/Menu.cpp.obj
[  3%] Building CXX object src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150
W/CMakeFiles/cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atme
ga32.dir/__/__/__/__/core/menus/Options.cpp.obj
C:/temp/cheali/src/core/menus/Options.cpp:34: warning: only initialized variable
s can be placed into program memory area
[  3%] Building CXX object src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150
W/CMakeFiles/cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atme
ga32.dir/__/__/__/__/core/menus/ProgramDataMenu.cpp.obj
C:/temp/cheali/src/core/menus/ProgramDataMenu.cpp:81: warning: only initialized
variables can be placed into program memory area
C:/temp/cheali/src/core/menus/ProgramDataMenu.cpp:90: warning: only initialized
variables can be placed into program memory area
C:/temp/cheali/src/core/menus/ProgramDataMenu.cpp:92: error: expected primary-ex
pression before '.' token
C:/temp/cheali/src/core/menus/ProgramDataMenu.cpp:92: error: expected primary-ex
pression before '.' token
make[2]: *** [src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150W/CMakeFiles/
cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atmega32.dir/__/_
_/__/__/core/menus/ProgramDataMenu.cpp.obj] Error 1
make[1]: *** [src/hardware/atmega32/targets/ADCKeyboardAnalyzer-150W/CMakeFiles/
cheali-charger-ADCKeyboardAnalyzer-150W_2.00-e10.3.12-20161002_atmega32.dir/all]
 Error 2
make: *** [all] Error 2

c:\temp\cheali>

```

stawel, 2016-10-02:
>Reinstal WinAVR to 20100110 (hope latest) and it started compiling.

hm... this WinAvr version uses avr-gcc version 4.3.2, it might be still to old,
maybe you should try to compile chaeli-charger with atmel studio?:
https://github.com/stawel/cheali-charger/blob/master/docs/building.md#atmega32---windows

I'm trying to do it myself, unfortunately my windows is updating  for two hours now :(
I also tried to compile it using cygwin but for some reason the avr packages disappeared.

stawel, 2016-10-02:
ok I was finely able to compile the project under windows:

1.(\*) install cmake https://cmake.org/download/ 
2.(\*) install make.exe from http://gnuwin32.sourceforge.net/packages/make.htm
3. (\*\*) install Atmel AVR Toolchain for Windows - avr 8bit toolchain http://www.atmel.com/tools/ATMELAVRTOOLCHAINFORWINDOWS.aspx
4.(\*\*\*) edit file https://github.com/stawel/cheali-charger/blob/master/cmake/FindChealiChargerInternal.cmake#L42
change: 
`#    set(name "${hardware}")`
to
`   set(name "${hardware}")`
5. compile:
(if you don't have make.exe in your path, see 1, but it seems you have it)
`c:\cheali-charger>set PATH=%PATH%;c:\Program Files (x86)\GnuWin32\bin`
compile:
```
c:\cheali-charger>set PATH=%PATH%;C:\avr8-gnu-toolchain\bin
c:\cheali-charger>cmake . -DCMAKE_TOOLCHAIN_FILE=avr-toolchain.cmake -G "Unix Makefiles"
c:\cheali-charger>make
```

(\*) - you don't need to do this, you have it already installed 
(\*\*) - you could install the whole atmel studio, but it much heavier
(\*\*\*) - I'm currently making a patch for this

Tomin123, 2016-10-02:
Prior of your last quide I downloaded AVR-GCC 4.9.2 (from http://andybrown.me.uk/2015/03/08/avr-gcc-492/)
and started compiling. It is ok at now.

I tried change to 35Volt in config file, but it seemes the same. :-(
In calibration-expert I can not change voltage on Vplus to anything from 35.000V
It behaves like this:
- enter to expert menu
- select line "Vplus:  6.974V"
- voltage "6.974V" starts blinking
- once I click up or down, it is redrawn to "35.000V" (and continue blinking)
- no change is possible
- if I exit by "STOP" button, it is returned to 6.974V
- if I confirm 35.000V, it changes to cca "35.002V"

Yesterday (before software change), once I confirmed 37Volts, then I could started charging without "error 4". (of course with bad voltage on display)
Today, even after full recalibration, start charging is not possible and shows new error: "calib. error, I charge 14"

stawel, 2016-10-02:
I'm not sure why you have:
> once I click up or down, it is redrawn to "35.000V" (and continue blinking)
I'm currently looking on it, but first try this:
1. reset eeprom, go to: "options"->"reset default"
2. calibrate voltage in  "options"->"voltage" (do not change anything in "expert DANGER!"
3. check if you are able to enter "charge"
if not, download your eeprom and send it to me.

Tomin123, 2016-10-02:
I did exactly what you wrote and in charge again "calibration error 4"
I did NOT calibrate any current and any expert, only voltage. Ok ?

In calib. menu I see perfect identical voltage in both V1-8 and Vout.
On pin 37 is 0.590Volt.
my eeprom is on www.fordiag.cz/download/tmp/chealieeprom.zip

stawel, 2016-10-02:
ok, based on your eeprom (your calibration) you should set:
#define MAX_CHARGE_V ANALOG_VOLT(29.700)
please check if this setting works.

Your charger should be able to measure 33.8V at least (4.2V*8cells), so either
1. your AREF voltage is to low (pin 29), I'm guessing it is currently ~2.32V, on most chargers it's 2.5V
or 
2. the AREF should be 2.32V by design and your output voltage divider is broken
or 
3. your charger doesn't use the AREF pin and the "internal 2.56V reference voltage" should be used
   (this requires a change in the firmware)

please check if the MAX_CHARGE_V value works, measure also your AREF voltage, and check if something is connected to this pin.

Tomin123, 2016-10-03:
Hi,

AREF is exactly 2.50Volt and it seems there is some vref IO source on pin 29 connected.
Checked it with scope and voltage is nice.

my premise: voltage divider on the output must be fine if I see in calib. menu V1-8 voltage exactly the same like Vout. (am I bad ???)
I have trimer in the divider (from manufacture) so I can set exactly the same voltage like in V1-8

Maybe there is a trouble in some divider(s) in V1-8 and it fools me to set Vout to bad point ?
Is it possible ? If so, what voltages should be after dividers (and MUXs) on A/D pins on the cpu ?
I have on outputs of OpAmps 1.87 (pin 14 of LM324) and 1.88Volts (pin 8)

I will try #define MAX_CHARGE_V ANALOG_VOLT(29.700)
Thanks again.

Tomin123, 2016-10-03:
New firmware with #define MAX_CHARGE_V ANALOG_VOLT(29.700)
after calibration, I got "calibration error", "I charge 14"
I am near to drop it and try to buy some another piece of China crap :-(

stawel, 2016-10-03:
> AREF is exactly 2.50Volt and it seems there is some vref IO source on pin 29 connected.
Checked it with scope and voltage is nice.

that's a surprise, based on your eeprom  the ADC on pin 37 reads a value of 16541
(value is left shifted, max value is 65504) for 7.560V input voltage,
so the maximum voltage should be 65504/16541*7.560V =~ 29.9V,
and you are measuring 0.590V on pin 37, so the AREF voltage should be  65504/16541*0.590V =~ 2.33V !?!?

> my premise: voltage divider on the output must be fine if I see in calib. menu V1-8 voltage exactly the same like Vout. (am I bad ???)
I have trimer in the divider (from manufacture) so I can set exactly the same voltage like in V1-8

in cheali-charger V1-8 and Vout is always the same after calibration, we just assume that a voltage
reading from the balance ports is equal Vout (after calibrating Vb1 to Vb8  we add the voltages and assign the sum to Vout)

> Maybe there is a trouble in some divider(s) in V1-8 and it fools me to set Vout to bad point ?
Is it possible ? If so, what voltages should be after dividers (and MUXs) on A/D pins on the cpu ?
I have on outputs of OpAmps 1.87 (pin 14 of LM324) and 1.88Volts (pin 8)

I don't think so, it doesn't really matter what the ADC reading on the balance port are , as long as 
the voltage is correct (the voltage you put in by hand into Vb1 and Vb2) and your eeprom 
clearly shows that you've put in 7.560V (to be exact you've set Vb1 to 3.778V and Vb2 to 3.782V)
and the Vout_plus_pin ADC measured at this point a value of 16541 (for the 7.560V)

>New firmware with #define MAX_CHARGE_V ANALOG_VOLT(29.700)
after calibration, I got "calibration error", "I charge 14" (I calibrated voltage and current, no expert)

that's actually  a good sign, we are one step further in the calibration check routine.
you need to decrees "settings"->"max Ic:"
https://github.com/stawel/cheali-charger/blob/master/docs/calibration_error_codes.md 
please set:
1. "settings"->"max Ic:"  to 4A
2. "settings"->"max Id:"  to 2A
and check if your charger works, then send me you eeprom again (after voltage, Ic and Id calibration)
I will check what the actual values are.

> I am near to drop it and try to buy some another piece of China crap :-(

;) bear in mind,  your effort will help others

Tomin123, 2016-10-03:
> in cheali-charger V1-8 and Vout is always the same after calibration

I did not have it, I had to change trimer value in input stage on pin 37 to achieve it.

> you need to decrees "settings"->"max Ic:"

Did it. Currently the same error: "Calibration error" "I charge 14"
My current eeprom is here www.fordiag.cz/download/tmp/chealieeprom20161003.zip

stawel, 2016-10-03:
this is how your calibration looks like:
```
const AnalogInputs::DefaultValues AnalogInputs::inputsP_[] PROGMEM = {

    {{0,  0},         {16574,  7534}},   //Vout_plus_pin
    {{0,  0},         {16574,  7534}},   //Vout_minus_pin
    {{1405,  50},         {7295,  1000}},   //Ismps
    {{575,  50},         {3136,  300}},   //Idischarge

    {{0,  0},         {44736,  23118}},   //VoutMux
    {{21500,  5200},         {41023,  2900}},   //Tintern
    {{0,  0},         {37439,  11011}},   //Vin
    {{6272,  2400},         {14300,  6000}},   //Textern

    {{0,  0},         {0,  0}},   //Vb0_pin
    {{0,  0},         {49218,  3764}},   //Vb1_pin
    {{0,  0},         {49319,  3770}},   //Vb2_pin
    {{0,  0},         {50280,  3837}},   //Vb3_pin
    {{0,  0},         {50602,  3862}},   //Vb4_pin
    {{0,  0},         {50752,  3865}},   //Vb5_pin
    {{0,  0},         {51008,  3885}},   //Vb6_pin

#if MAX_BANANCE_CELLS > 6
    {{0,  0},         {50752,  3865}},   //Vb7_pin
    {{0,  0},         {51008,  3885}},   //Vb8_pin
#endif

    {{8934,  50},         {22650,  1000}},   //IsmpsSet
    {{231,  50},         {1735,  300}},   //IdischargeSet
};
```

this is what the ADC sees:
` {{1405,  50},         {7295,  1000}},   //Ismps`
we are reading 1405 for 50mA and 7295 for 1A, this is definitely not correct,
the first value should be about 20x smaller.
 
and this is the PWM we need to set:
`{{8934,  50},         {22650,  1000}},   //IsmpsSet`
we need to set the PWM to 8934 to achieve a 50mA, and 22650 for 1A.
(also questionable)

if you do have a different multimeter please repeat the calibration with it.
you can also try to change the "Iexpect" from 50mA to 100mA maybe this will help 
(at the bottom in "calibrate"->"I charge"->"50mA" menu), if you do this next you need to set the 
"value" so that your multimeter shows also 100mA.

Tomin123, 2016-10-04:
Hi Pawel.
Heureka ! :-)
Calibration of current was very unstable. Sometimes I calib. 100mA and then went to 1000mA.
Calib. 1000 and then check again 100 and it draws only 45mA, for example.
So I started discovering and measuring and realize that there unstable "5 2/1" OpAmp.
There is a capacitor in feedback of this OpAmp and this was bad.
On the schema there 220n, but my cap. was 2M5. With new 200n it was even more unstable and
start oscilate. So I replace it to 2M5 electrolytic and now it is a "new charger".
Very stable, current displays max 2mA error, no oscilation.

Many thanks for your help again. Superb support.
Good luck.
Tomas

my schema with remarks and colored bad capacitor (for other wretches) is here
[image]
[image]