stawel/cheali-charger #200: Accucel 8 150 slow charging

DieHertz, 2016-10-03:
I have flashed my Accucel 8 150W to the latest stable firmware, performed voltage calibration, although no current calibration. When I attempt to charge my batteries, the charging current won't go above 188mA. Batteries I'm trying to charge are 1800mAh 2S and 1800mAh 4S.
I understand that I have likely configured something incorrectly, but maybe you'll be able to help.

stawel, 2016-10-03:
you have to do the current calibration first, maybe then I may be able to say something more.

DieHertz, 2016-10-03:
It may be non-functional unless current calibration is performed? I'll try to get current clamps this week, maybe calibration will solve my problems.
Thank you, Pawel.

stawel, 2016-10-03:
there might be also a second reason, if you batteries are in a bad condition (during the 188mA charge the battery voltage significant changes) the charger won't increase the charging current (until voltage stabilizes)

stawel, 2016-10-03:
when you do the calibration could you send me your eeprom?

DieHertz, 2016-10-03:
@stawel Sure, will do.

stawel, 2016-10-03:
hm... I've check the default calibration for your charger, it's definitely wrong in the "I charge"  and "I discharge" section, so without calibration it won't work correctly.

DieHertz, 2016-10-04:
After calibrating with an ampermeter the process seems close to reality, although the value indicated during the constant current phase slightly differs from the *Ic* for the battery (Ic = 1800mA, indicated value around 1795).
Will post the EEprom image tomorrow!

DieHertz, 2016-10-06:
Apologies for the delay.
Here's my EEprom file. The ampermeter wasn't particularly precise, but I've ensured that it shows 0.05 and 1.0 when charging current is 50mA and 1000mA respectively. Also calibrated discharge current and voltages.
[eeprom.hex.txt](https://github.com/stawel/cheali-charger/files/514466/eeprom.hex.txt)

stawel, 2016-10-06:
thanks!
I've added new default calibration based on your eeprom.

>  although the value indicated during the constant current phase slightly differs from the Ic for the battery (Ic = 1800mA, indicated value around 1795 mA).

you can try to increase the accuracy by setting other calibration points, for example 100mA and 2000mA,
for that you need to:
1. change the "Iexpect" from 50mA to 100mA 
(at the bottom in "calibrate"->"I charge"->"50mA" menu)
2. next you need to set the "value" so that your multimeter shows also 100mA. 
3. do 1.,2. for the second point  (1000mA),  this time set "Iexpect" to 2000mA

DieHertz, 2016-10-07:
Glad to be helpful.
Thanks, the issue can be closed now 👍

closed 2016-10-07 by DieHertz