stawel/cheali-charger #205: Very weak buzzer sound

neosunrise, 2016-11-07:
I have managed to run the cheali charger firmware on my Dynamite Passport charger and now everything seems good but the buzzer sound.

With the stock firmware, the sound was really clear and loud which I like. However, with the cheali charger firmware, the buzzer sounds really weak and the tone of the alarm when charging is complete sounds really low and scary. How can we fix it so that it could be the same as the stock firmware?

Thank you!

neosunrise, 2016-11-08:
btw, I think the sound in version 0.33 was not bad, at least better than the current version. I found that their implementations are different.

stawel, 2016-11-08:
Yes, the buzzer implementation is quit poor (never had high priority).

the buzzers PWM implementation (frequency) is here:
https://github.com/stawel/cheali-charger/blob/develop/src/hardware/atmega32/cpu/Timer0.cpp#L58-L69

the Buzzers modulation is implemented in:
https://github.com/stawel/cheali-charger/blob/develop/src/core/drivers/Buzzer.cpp

Note also that I'm showing you the "develop" branch,
if there will ever be a next release (I'm not sure if I will find time for that)
then it will be from this branch.
If you are interested in adding you code to the project you should start from this branch
(not every patch will be added, mostly because of lack of free program memory)
of course you can always make your own branch of the entire project.

neosunrise, 2016-11-08:
Thanks for the info. Can you by any chance let me know how to change the frequency? Also like I said, the buzzer sound in 0.33 version is not bad but it looks like that you took out everything related to the buzzer from imaxB6.cpp to timer0.cpp. I believe you also made some other changes. This is why your replies to other people with the same concern do not really help in the newer version source code. Would it be even possible to make the buzzer buzz like the stock firmware?

neosunrise, 2016-11-10:
Pawel, I am writing a Chinese manual for this charger and would like to give credit to you ( and Cheali charger of course). May I know if you are the only original author of this firmware? Do you mind me making some changes to the firmware and have my forum ID (neosunrise) on the booting screen and distribute it to the Chinese hobbyists?

stawel, 2016-11-10:
>  Also like I said, the buzzer sound in 0.33 version is not bad but it looks like that you took out everything related to the buzzer from imaxB6.cpp to timer0.cpp. I believe you also made some other changes.

Yes, currently the code uses a timer for the buzzer, mainly because I wanted to add a frequency setting 
and maybe a volume setting (not sure If the volume setting would actually work), but as usually I didn't found the time.

The documentation for timer0 you can find in:
http://www.atmel.com/Images/doc2503.pdf
page 69.
we use the "Fast PWM Mode" page 75 (of course we can use any mode).

the important code is here:
https://github.com/stawel/cheali-charger/blob/develop/src/hardware/atmega32/cpu/Timer0.cpp#L64-L66 

The counter is counting from sound_TCNT0 to MAX (256),
we change the buzzers pin to "1" when counter reaches OCR0,
currently it is in the middle of the count, so:
frequency  ~ (is linear proportional to) MAX -  sound_TCNT0 = 256 - sound_TCNT0
duty = 50% =(OCR0 - sound_TCNT0)/ (MAX -  sound_TCNT0)

to change the frequency you have to change sound_TCNT0,
to change the duty (volume) you have to set  OCR0 = sound_TCNT0 + something,
maximum OCR0 value should not exceed OCR0 = (sound_TCNT0 + 256)/2 == 50% duty.

> Pawel, I am writing a Chinese manual for this charger

This is a great initiative!

>and would like to give credit to you ( and Cheali charger of course). May I know if you are the only original author of this firmware?

No I'm not the only author, there are many people who wrote code and helped in the development,
some of them are listened here: 
https://github.com/stawel/cheali-charger/graphs/contributors
https://github.com/sasam/M0517_flash_tools/graphs/contributors
there are also many forks which where inspiration for this code:
https://github.com/stawel/cheali-charger/network
And of course the most important people (in my opinion)
are those who write documentation and schematics
 (see https://github.com/stawel/cheali-charger#useful-materials)
and answer questions on issues (github)
and on cheali-charger (at) googlegroups.com

>Do you mind me making some changes to the firmware and have my forum ID (neosunrise) on the booting screen and distribute it to Chinese hobbyists?

I even insist on this, especially when you do a fork,  
you can even change the firmware name if you like,
of course your fork needs to be open source too.

neosunrise, 2016-11-11:
Sounds good. Thank you!