stawel/cheali-charger #206: Turnigy Max80W small problem

mrzed72, 2016-11-21:
Dear Pawel!

Very good job, thank you.
There is only a small problem:
Only I can spend ~45W, ~80W does not work.
If it is not too much work, please look for the cause of the problem.
Sorry bad english, i am use google translate.

Thank you
Zoltan.

stawel, 2016-11-21:
> Only I can spend ~45W, ~80W does not work.

1. are you using: cheali-charger-Turnigy-MAX80W-7A_2.00-e10.3.12-20160613_atmega32.hex (or .bin) ?
2. what is the output current Ic?
3. what is the output voltage Vout? (charging voltage)
4. what is the balance port voltage? (if used)
5. what program are You using: charge, discharge, balance+charge,...?
6. which battery type?

mrzed72, 2016-11-22:
1. I flash cheali-charger-Turnigy-MAX80W-7A_2.00-e10.3.12-20160613_atmega32.hex
2. Set 5000mAh to 8A 6S lipo battery. 2 to 1.8mAh charging steady decline
3. 22,2 to 25.2 V
4. 3.7 to 4.2
5. charge or balance + charge
6. 8000 mah 6S Li-Po

It recharges the battery properly, but the power does not exceed 45W was never.
If settings menu max IC > 5000mAh then "calib. error I charge 14".

stawel, 2016-11-22:
I've uploaded a new version for you:
cheali-charger-Turnigy-MAX80W-7A_2.00-e10.3.12-20161122_atmega32.hex

please check if it work now.

mrzed72, 2016-11-23:
I flash cheali-charger-Turnigy-MAX80W-7A_2.00-e10.3.12-20161122_atmega32.hex
I calibrated and tested.
Unfortunately, still it does not work correctly:
https://youtu.be/F0iMhJBNp-Y

stawel, 2016-11-23:
Thanks for the video!
could you do another one, but this time run the "charge" program,
I also need to  see the cells voltage during charge (press the "up" button several times while charging).

mrzed72, 2016-11-23:
Ready:
https://youtu.be/NIjHtJ9f-qk
Input voltage not valid, i use pc power supply max 12V output.

stawel, 2016-11-23:
hm... interesting, I have no idea what's going on :/, i need to think it through,
could you also download your eeprom from your charger and send it to me?

mrzed72, 2016-11-23:
did you think of this?
https://drive.google.com/file/d/0B-1Wi6VTV4CvclFoRUdwdnhIMjg/view

stawel, 2016-11-24:
>did you think of this?

Yes, thank you very much,
based on you eeprom I can say that your "settings"->"max Ic" should be ~ 6700mA
but I still don't know why you can't charge with power > 45W :/
I will look into it after this weekend - unfortunately I'm busy until then.

if you have a battery with less cells (<6cells) could you check if you can charge it with >45W?

stawel, 2016-11-24:
> Input voltage not valid, i use pc power supply max 12V output.

why is this voltage not valid?
could you calibrate "Vin" and check what the voltage is during charge?

njozsef, 2016-11-24:
Szia Zoli.

Ha az a problémád, hogy nem tudod a max áramot kihasználni, az azért van,
mert
az áramot mérő sönt túl nagy a töltőben (az eredeti szoftver is rossz volt
szerintem).

Szóval le kellene ezt a mérőellenállást csökkenteni és utána az áramot
megpróbálni újrakalibrálni.
A hibát az okozza, hogy a vezérlés növekedésekor túlcsordul az az aránypár
ami a töltőáram/mérési adat-ot reprezentálja.

Sorry guys.

This is a hardvare problem, need change smallest current measure resistor.
best regrds
nj

mrzed72, 2016-11-30:
stawel:
I'm sorry. I did not know.
Not available in the original firmware?

stawel, 2016-12-02:
@mrzed72  could you test the new firmware cheali-charger-Turnigy-MAX80W-7A_2.00-e10.3.12-20161202?

as @njozsef said, the  "settings"->"max Ic" issue is a matter of your current measuring resistor,
you need to set "max Ic" to something less than  6700mA or replace the resistor.

mrzed72, 2016-12-03:
I flash and calibrating... tested...
It became a little better:
https://youtu.be/gEwd3lx99sc
The same more 3S in parallel with batteries.

stawel, 2016-12-09:
> It became a little better:

I'm suspecting that your issue has something to do with this setting:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/Turnigy-MAX80W-7A/HardwareConfig.h#L45

This value sets a PWM duty limit on the boots converter,
1.56 means a 56% duty limit.
(see: https://en.wikipedia.org/wiki/Boost_converter#Continuous_mode)
If you what you can try to increase it to 1.60 and compile the code.

> The same more 3S in parallel with batteries.

I'm not sure if I understand you...
what is the power output when you connect a 3 cell battery?