stawel/cheali-charger #207: New M0517 Hardware  V1.1 2016.06.25

Ralim, 2016-11-28:
Hi,
I love the firmware and currently run it on quite a few of my chargers using the atmega chipset.
I ordered this new charger and it came with the number marking removed but checking pinout for power and XTAL showed it as a match for the M0517. I have traced the programming to the J8 header at the bottom of the unit [GND -x -x - SWCLK - SWDIO]

Attached is a photo of the board.

I have tried all the firmware in the repository and none of them work (nothing on the LCD).

Therefore I believe this unit must use a different pinout.
I am currently attempting to trace out the connections to the chip to the hardware. So far it looks like its definitely a different pinout.

If no-one has any details on this new device I'll trace connections and post back :)

[Images](http://imgur.com/a/V8cez)

diggit, 2016-11-28:
Hi @Ralim,
if you can confirm matching xtal, vcc, gnd (all) and programming pins with M0517, there is possibility of real M0517 sitting there. If not, it is some unidentified/unsupported MCU...
If mentioned pins match, it should be possible to identify chip using swd. Did swd programmer show some information?

I saw 4051 on your PCB. That is kinda suspicious. AFAIK M0517 has enough ADC channels and does not need any external analog multiplexer.

Ralim, 2016-11-28:
Hi,
I can confirm it is a M0517 as it responds to being programmed via the st-link programmer.
All the power, xtal and reset lines match as well.

I have traced out most of the pins on the chip & their functions...
Here is what i have currently traced out :
```

Pin -> function
1  -> 
2  -> 
3  -> 
4  -> RST -> pin 4 of J6 (has trace to cap)
5  -> J8 pin 3 UART RX
6  -> GND
7  -> J8 Pin 2 UART TX
8  -> 4051 MUX CTRL A
9  -> 4051 MUX CTRL B
10 -> LCD DB4
11 -> LCD DB5
12 -> 4051 MUX CTRL C

13 -> LCD DB6
14 -> LCd DB7
15 -> XTAL (12 Mhz)
16 -> XTAL
17 -> GND
18 -> CAP to GND
19 -> LCD Enable
20 -> LCD RS
21 -> 
22 -> a transistor on the +ve input of lm393 opamp near dc in
23 -> discharge control
24 -> 

25 -> taps singal from lm393 in upper left of board
26 -> 1K resistor to pin 40
27 -> disconnected?
28 -> left button
29 -> 2nd from left button
30 -> 3rd from left button
31 -> 4th from left button
32 -> Buzzer drive
33 -> goes to R45 (not populated drive for a fan)
34 -> Cell 2 discharge mosfet
35 -> Cell 1 discharge mosfet
36 -> 

37 -> Cell 3 discharge mosfet
38 -> Cell 4 discharge mosfet
39 -> Cell 5 discharge mosfet
40 -> Cell 6 discharge mosfet
41 -> VDD
42 -> VDD
43 -> 17 k to batt -ve 
44 ->
45 -> 
46 -> Current shunt reading input
47 -> 4051 MUX output via 300 ohm resistor
48 -> shuts down signal to charge mosfet
```

Not sure of some of the pins though :/

diggit, 2016-11-28:
That looks good!
Even M0517 designs uses external mux as I discovered now.
[Look into this file.](https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/imaxB6-pins.h) You should trace all pins described there. [Schematics can help you](https://github.com/stawel/cheali-charger#useful-materials).
You can modify that file, build FW and test it. If you are unsure about anything, be sure to ask. If you have no build environment, I can build modified FW for you.

Where did you get this charger?

Ralim, 2016-11-28:
Awesome, ill keep tracing it out :) ..Trying not to get confused.

Bought it from [hobbyking](https://hobbyking.com/en_us/imax-b6-dc-charger-5a-50w-copy.html) about a week ago. So who knows where they sourced it from.

I try and modify and build and test and let you know :)

Ralim, 2016-11-28:
Some progress:
Working:
* LCD
* Buttons
* Cell readings

Not working:
* charging / discharging

[Current pin asignments](http://pastebin.com/5jyXmSbM)

If anyone has any ideas of where to look tomorrow ill have a poke around :)
Current plan is to scope probe some of the signals and see whats not working correctly

diggit, 2016-11-28:
Wow, nice progress.
To get charging working, you'll need all analog value readings.

Then you need all enable/disable pins correctly mapped.
I am not sure, if PWM pin remapping is enough or  more changes are needed.

@stawel does buck, boost and discharge PWM pins need more config than just remapping? Like setting up different timer channels or so...

stawel, 2016-11-28:
> @stawel does buck, boost and discharge PWM pins need more config than just remapping? Like setting up different timer channels or so...

hm... only PWM pins 20 and 26 are implemented right now:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/outputPWM.cpp#L97-L116

stawel, 2016-11-28:
@Ralim  before starting any charging/discharging please make sure all ADC inputs are correctly assigned,
you may find the AnalogInputsAnalyzer useful:
https://github.com/stawel/cheali-charger/tree/master/src/hardware/nuvoton-M0517/targets/AnalogInputsAnalyzer-50W

it should show you all ADC values from all inputs,
(these is also a dedicated firmware version for checking the balance port LOAD pins).

Personally I would first try to start discharging, for that you need at lease:
ADC inputs:
1. OUTPUT_VOLTAGE_MINUS_PIN
2. OUTPUT_VOLTAGE_PLUS_PIN
3. DISCHARGE_CURRENT_PIN
other pins:
4. OUTPUT_DISABLE_PIN (disconnect the "-" main lead wire, should be set to LOW when discharging)
5. DISCHARGE_VALUE_PIN (discharge PWM, may be common with SMPS_VALUE_BOOST_PIN)
6. DISCHARGE_DISABLE_PIN (disable discharge,  should be set to LOW when discharging)
7. SMPS_DISABLE_PIN (disable charge,  should be set to HIGH! when discharging or you burn something)

Also a good practice is to connect an old type light bulb (low Ohms high power resistor) in series to the battery and a second one to the power supply - this reduces the chance of burning something.

Ralim, 2016-11-28:
Hi, 
Thank you for helping :) 

I have been flying with the input analyser firmware as well,  it's really handy :) 

I have correct readings coming in for 
Vin, B+ and all the cell voltages. 

I have the correct pin for the current shunt for discharge,  however the unit I have has a gain of 10 instead 3 that I see in most of the schematics.  Not sure if that will cause any issues or not. 

Thurs unit does have the discharge pwm common to the boost pwm,  so that part is similar. 

One big issue I have found is that this unit does not appear to have an adc connection to B- directly,  and the one one I can find is via the balance plug feedback circuitry (they are tied together by 2k ohms) 

The other small issue is the unit uses a 12 mhz crystal,  so later on is there somewhere where I should define the new f_cpu? 

What firmware build do I use to check balance port pins? 

I'm currently using a bench supply for the units power with current limiting set,  I'll add a few ohms inline to the battery :) great idea. 

I can confirm the B- mosfet is being switched on correctly and does join the B- to system ground when I enter discharge mode and also disconnects when leaving discharge mode.

stawel, 2016-11-28:
>I have the correct pin for the current shunt for discharge, however the unit I have has a gain of 10 instead 3 that I see in most of the schematics. Not sure if that will cause any issues or not.

shouldn't  be a problem, current calibration should handle it.

> One big issue I have found is that this unit does not appear to have an adc connection to B- directly, and the one one I can find is via the balance plug feedback circuitry (they are tied together by 2k ohms)

hm... the B- ADC reading is quite important (should be somewhere), you could connect a battery in reverse (with an resistor in series of course) and look which ADC input reacts the most
(the B- input  determines when a battery is connected the wrong way).

> The other small issue is the unit uses a 12 mhz crystal, so later on is there somewhere where I should define the new f_cpu?

we initialize the clock in:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/cpu/cpu.cpp
the m0517 cpu is design to work with 12MHz not 16MHz (at least the CMSIS is written so)
so we had to change also:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/cpu/CMSIS/Device/Nuvoton/M051Series/Include/system_M051Series.h#L39-L40

in the future we probably should  add a parameter to target/*/CMakeLists.txt

>What firmware build do I use to check balance port pins?

it's currently not added to the m0517 architecture, you could add it or you could just change:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/targets/AnalogInputsAnalyzer-50W/HardwareConfig.h#L25
to
#define ENABLE_HELPER_BALANCE_PORT_ANALYZER

Ralim, 2016-11-29:
Hi,
using the two test firmware i have confirmed the following :

- Balance sense connections all read correctly (well up to 5s don't have a 6s currently to test with yet)
- Balancing resistor drives all work (up to 5s)
- VIN+ measurement works
- B+ measurement
- B- sort-of
- OUTPUT_DISABLE_PIN
- SMPS_DISABLE_PIN
- DISCHARGE_DISABLE_PIN

However, I am not getting any PWM out of the discharge PWM pin (which is shared with BOOST).
I assume this is because this is on pin 23 instead of pin 20 in the original designs.
Also, BUCK drive is on pin 25, so it seems this unit has different pins for both of these drive signals :(

I will try and look into gaining PWM output on these two extra pins next :)

Ralim, 2016-11-30:
Progress!

I now have discharge working, I have edited the outputPWM.cpp file for the settings for the different pins :)

Discharge PWM is working, though i think i damaged something in testing.. Currently investigating :/ ... oops

Ralim, 2016-12-13:
Hello,
Good News @stawel.
I have working files for this unit now (charging/discharging/balancing working and tested).
I have noticed when using an oscilloscope that the PWM output is running at 25Khz rather than 32Khz. Will this pose an issue for long-term operation of the unit?

Otherwise, the only issue stopping me opening a pull request is I have no idea how to pull off the eeprom settings to be able to fill out the default settings file for the unit?

Thank you for your great software :)

stawel, 2016-12-13:
congratulations!

> I have noticed when using an oscilloscope that the PWM output is running at 25Khz rather than 32Khz.

you can fix this by looking at the "pwm_n" value in:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/outputPWM.cpp#L85-L86
and set it into:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/outputPWM.h#L22

if you know how to run the firmware in debug mode you could just set a breakpoint at line 86 and read 
"pwm_n" through a debugger,
OR
if your UART is working you could:
uncomment line:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/outputPWM.cpp#L23
and change line:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/outputPWM.cpp#L99
to:
LogDebug("my pwm_n is:", pwm_n);

> Otherwise, the only issue stopping me opening a pull request is I have no idea how to pull off the eeprom settings to be able to fill out the default settings file for the unit?

just send me a eeprom.bin and I will add it myself,
normally you do this by running in linux:
```
$ cd cheali-charger/utils/eepromExtractor
cheali-charger/utils/eepromExtractor$ ./eeprom.py eeprom.bin
```
This should create a defaultCalibration.cpp file, but you need some additional python packages to run  "./eeprom.py".

diggit, 2017-02-05:
Hi @Ralim, any progress here? We'll be happy to have new board supported.

Ralim, 2017-02-06:
Hi, Sorry about the mega delay.
Currently running into two intermittant issues:
1. The cell voltage readings dont go below 2V (ie connected to battery they read correctly, but when battery disconnected they do not go down to zero).
2. The PWM output does not turn on for charging sometimes for the high side FET (ie output voltage below input).

Currently I'm poking away in my spare time, but starting to run out of ideas for both of these.
Not sure why they both seem to be new issues as well that appeared after a re-flash of previously working file (no recompile either)...

Im happy to keep plodding along, unless you have some amazing ideas :)

diggit, 2017-02-06:
>1. The cell voltage readings dont go below 2V (ie connected to battery they read correctly, but when battery disconnected they do not go down to zero).

What happens when you short some input of ballancer? Does it go lower? Battery is disconnected from balancer or charger at all? Non zero readout is ok, but should be below 0.5V.

>2. The PWM output does not turn on for charging **sometimes** for the high side FET (ie output voltage below input).

Did you try to look on gate with oscilloscope? In case Uin>Uout, high side (buck)transistor should have PWM on gate.  If Uin<Uout, high side FET should be turned on and low side(boost) FET should have PWM on gate.
That "**sometimes**" sounds bad, non consistent errors are not easy to hunt.
Can you try to find exact situation when it happens?

@stawel any ideas?

universam1, 2017-02-07:
Thanks @Ralim @stawel 
I have the same one from Hobbyking and could assist with testings.

Ralim, 2017-02-10:
@diggit 
Sorry to be coming back with mixed news.. I think I have narrowed down whats causing me most of the issues.
I loaded the hardware analyser back onto the unit, and all the vb1-vb6 pins read really high when disconnected (45000~) and when connected to the battery they read 65520 which makes sense as a max value.

The actual voltage on the ADC input is around 1.7V.
What is the input range of the ADC itself? (is it a 1.5V input that is being overloaded)?

I'm currently out of ideas for how to diagnose the ADC readings at the moment.

diggit, 2017-02-11:
@Ralim 
According to datasheet, there is no internal ADC reference and cheali config does not mention external ref. either. This means voltage range of ADC is probably GND ~ Vcc (3.3V).

(I expect, that you have all pins properly mapped.)

In [AnalogInputsADC.cpp](https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/AnalogInputsADC.cpp) is whole ADC process. There is some input capacitor discharge necessary. Do you have main XTAL frequency same as other M017 chargers? If it is higher, there might be not enough time to discharge this cap. Just guessing...

What does ADC measure when you short 2 neighboring balancer pins? Are other ADC input values affected?

I am not much familiar with insides of this FW. I believe @stawel should have better idea what is wrong.

stawel, 2017-02-24:
> I believe @stawel should have better idea what is wrong.

unfortunately I don't :/

maybe You should try something like this, add at line:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/AnalogInputsADC.cpp#L208

```
_setMuxAddress(0); //test values for 0 to 7
while(1) {}
```
this should set the multiplexer address and hang the code after ADC initialization,
then check the ADC pin voltage (should be ~0V). Check voltages for all mux addresses for 0 to 7.

Also You could commit your code to github, maybe I will find the issue in code.

The only  ideas that comes to my mind:
1. hardware is broken
2. my code has some bugs (unfortunately this is very likely)
3. the pin is also somewhere else used as a digital pin.

ps.
Sorry for the long delay, I just started a new job and the only thing a currently do in my spare time is sleeping :)

Ralim, 2017-03-16:
Hi, Sorry for the delay, my main computer is currently out of action (Dead Motherboard).
I will try the manually setting the MUX when I get a machine to build on.
I have pushed the code up to my fork https://github.com/Ralim/cheali-charger when I was last working on it.

I don't think issues 1 or 2 are occurring, however, I will check as shortly.

closed 2018-10-31 by Ralim