stawel/cheali-charger #208: imax b6ac

DenisZavodsky, 2016-12-04:
hi 
i need help. just re flashed my imax charger and all i get is squares in first rov and some beeps  but nothing else  my hardvare can be seen in the picture attached 
[image] i flashed it with 
 cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin  file using awrduddes 2,4
any one can help what i doing wrong?

DenisZavodsky, 2016-12-04:
[image]
pic of the charger

diggit, 2016-12-04:
Hi @DenisZavodsky 
Look into [Readme troubleshooting sections](https://github.com/stawel/cheali-charger#troubleshooting).
 > __download the \*.hex again, use the "RAW" button in github__

Most common problem is directly downloading from link in github directory view.
What was size of you **.bin** file? It must be below 32768B (32kiB). (hex will be larger)

Post here avrdude output if this does not solve your problem.

DenisZavodsky, 2016-12-05:
diggit i did set fuses but no joy here is the screen shot +output from avr dudess
[image]    Detected 1e9502 = ATmega32
~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ 

avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware update.
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9502
avrdude.exe: reading lfuse memory:

Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "C:\Users\DENIS\AppData\Local\Temp\1beeedc1-8a01-4b10-a911-4d7d45163bb7.TMP"
avrdude.exe: reading hfuse memory:

Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "C:\Users\DENIS\AppData\Local\Temp\ce23e078-856b-4bb1-97b0-008de0a22616.TMP"
"efuse" memory type not defined for part "ATmega32"

avrdude.exe done.  Thank you.

~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ 

avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware update.
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9502
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware update.
avrdude.exe: reading input file "C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin"
avrdude.exe: input file C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin auto detected as raw binary
avrdude.exe: writing flash (31738 bytes):

Writing | ################################################## | 100% 23.78s

avrdude.exe: 31738 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin:
avrdude.exe: load data flash data from input file C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin:
avrdude.exe: input file C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin auto detected as raw binary
avrdude.exe: input file C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin contains 31738 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 16.37s

avrdude.exe: verifying ...
avrdude.exe: 31738 bytes of flash verified

avrdude.exe done.  Thank you.

DenisZavodsky, 2016-12-05:
and avr comand 
-c usbasp -p m32 -B 0.5 -U flash:w:"C:\Users\DENIS\Desktop\cheli\cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin":a

diggit, 2016-12-08:
@DenisZavodsky 
Are your fuses correctly set (readme)? Can you try to flash your charger with hex file? Just for sure.

@stawel any idea what could be wrong? I hope, that MCU pin usage is identical, but what else can cause this?

DenisZavodsky, 2016-12-08:
solved
 in avr dudess had to uncheck the JTAG Interface Enabled in fuse settings and it started to work as it should 
thank you all for help

diggit, 2016-12-09:
Great!
AFAIK suggested  fuses in readme disables JTAG by default.

diggit, 2017-07-04:
Seems to be solved, can you close this issue?

closed 2017-07-15 by DenisZavodsky