stawel/cheali-charger #209: IMAX B6 (original): Wrong UART bitrates at high speeds (38400++)

S-trace, 2016-12-06:
I'm have flashed original IMAX B6 with stawel/cheali-charger@fe5e5c0a and have issues with PC connection. It works fine on 9600 and 19200, but on 38400++ only garbage is recieved.
I have logic analyzer device and so decided to measure real UART bitrates on adapter and IMAX, and found werid thing: 
Adapter bitrates (nominal vs real): 
9600: 9933
19200: 19277
38400: 39538
57600: 57692
115200: 115384

IMAX bitrates (nominal vs real): 
9600: 9884 (works with adapter)
19200: 20304 (works with adapter)
38400: 43087 (garbage only with adapter, framing errors with analyzer)
57600: 70381 (garbage only with adapter, framing errors with analyzer)
115200: 175182 (garbage only with adapter, framing errors with analyzer)

Is there something bad with mine IMAX?
I had tried to use another adapter, but it doesn't worked at all (no garbage even).

diggit, 2016-12-06:
Hi @S-trace,
Your charger crystal oscillator seems to be oscillating on different frequency. it has 112% baud rate on 38400. That is too much. (never seen it)
You can measure frequency on crystal if you have oscilloscope. There should be 16MHz.

I can't calculate it's frequency from prescaler formula. Did your measurements matched repeatedly?

Can you upload zip somewhere else? Github reports error when I try to download it.

diggit, 2016-12-06:
Can you capture several frames with higher sample rate? 16MHz or more if possible.

S-trace, 2016-12-06:
[IMAX_B6_UART_115200_v2.logicdata.zip](https://github.com/stawel/cheali-charger/files/635016/IMAX_B6_UART_115200_v2.logicdata.zip)

diggit, 2016-12-08:
Still 2.4MHz sample rate.
Shortest pulse seems to be around 5.8us, It should be around 8.7us fo 115200. That is strange. Can you check value printed on crystal?

diggit, 2016-12-08:
Also you need your uart to be set to 8 data bits, no parity, 1 stop bit. No sure what do sou use for data receiving.

diggit, 2017-07-04:
Hi @S-trace, did you solve this issue?

S-trace, 2017-07-04:
Hello, @diggit 
No, i did not solved it. The oscillator in mine IMAX have 16.000 printed on its cover, but i'm have no oscilloscope to measure real frequency.
I'm always using 8n1 options for UART adapter with IMAX, bitrate is only variable value.

diggit, 2017-07-04:
You can try to slightly increase F_CPU in [avr-compiler.cmake](https://github.com/stawel/cheali-charger/blob/master/avr-compiler.cmake#L13) and build new FW with this value changed.

stawel, 2017-07-08:
if you are using a CP2102 based UART dongle you could also try to remove a 1kOhm resistor:
https://groups.google.com/forum/#!searchin/cheali-charger/uart|sort:relevance/cheali-charger/LKy6qITaVDg/FvYe_ZY7C9EJ