stawel/cheali-charger #210: Hobbyking HKC6

nickxab, 2017-01-10:
Can i install this FW in Hobbyking HKC6 charger ?

stawel, 2017-01-11:
I don't really know, do you have pictures of it's PCB?
what CPU is inside?

nickxab, 2017-01-11:
Here is my charger: http://imgur.com/a/1BRar
Sadly the cpu is scratched from the upper side from the factory so i can't see the model.
Is a "copy" of Turnigy Accucell 6 if i'm not mistaken. I followed this mod to add the USB data logging on my charger: http://reddishrock.blogspot.gr/2013/08/turnigy-accucell-6-mod-1-adding-usb.html
Is there any way to check the cpu model using my pc ?

stawel, 2017-01-12:
>Sadly the cpu is scratched from the upper side from the factory so i can't see the model.

my guess is that is is not supported - most of the chargers without a label on CPU are  presumably Z51F3220FNX chips which are not supported.

> Is there any way to check the cpu model using my pc ?
1. you could check if the quartz is 16MHz (atmega32 base boards  use such a crystal)
2. check if the pins have something in common with atmega32 http://www.atmel.com/Images/doc2503.pdf page 2, especially pins:
a) MOSI,MISO, SCK, ~RESET - programmer pins
b) VCC, GND
c) XTAL1, XTAL2 (quartz crystal pins)
3.  if you have a avr ISP programmer you could check if avrdude connects to (recognizes) the cpu - but please check first how the 10pins connector is connected to the CPU.

nickxab, 2017-01-12:
The pinout is exactly like atmega32, my crystal is also 16MHz.
http://imgur.com/a/FKKvZ

stawel, 2017-01-12:
maybe it is a atmega32, have you tried to flash it?

nickxab, 2017-01-13:
I don't have my AVR programmers here :/ but i have an Arduino UNO here. As i know you can flash them using arduino...should i give it a try ? :D

stawel, 2017-01-15:
It's up to You, but be aware, I have never seen such a PCB so there is a change we will have to do some
modifications to the firmware.

nickxab, 2017-02-25:
i have my usbasp now and i used the eXtreme Burner- AVR program on windows to verify my chip, indeed is atmega32 (i'm lucky). That charger is 60W by the way.
What can i do now ? I mean are 60W chargers supported ? :)

stawel, 2017-02-25:
>  ...i used the eXtreme Burner...

please don't touch the avr fuses, or you will most likely brick your charger ;)
in my opinion avrdude is much safer to use but it's a console ("cmd.exe") program.

It would be wise to check if other CPU pins are connected like in a imax B6 charger:
http://www.rcgroups.com/forums/showatt.php?s=df7049bcbafdb5d7d06765c264e5c4bb&attachmentid=3693125&d=1293732709
(if the PCB layout is the same)

then you could try this version:
https://github.com/stawel/cheali-charger/blob/master/hex/cheali-charger-imaxB6-original_2.00-e10.3.12-20160613_atmega32.hex

if it works I will make a 60W version for you.

nickxab, 2017-02-26:
Oh i see, thank you very much for the response. :)
I will open it again and will check them.
Avrdude is this one, right ? http://www.nongnu.org/avrdude/
One more question, can i back up the stock FM from my chip ? so just in case that something goes wrong i can restore it back.

diggit, 2017-02-26:
>...can i back up the stock FM from my chip ?

[Look here.](https://groups.google.com/d/msg/cheali-charger/7EG04N4jQf0/6Tl2OgeuAAAJ)