stawel/cheali-charger #211: auto Pb (Ca/Ca technology) 16,5V battery

niarbagem, 2017-01-13:
Good day!
I had founded some information (on automotive forums) about "Pb (Ca/Ca technology) battery" that charging end voltage is 16,2 - 16,5V.
Is it possible to make manual "overcharge" for Pb battery up to 3v per cell?
Accucell-8 150W cheali-charger v.2.0

diggit, 2017-01-13:
Hi,
It is possible. In settings, set menus to advanced and then edit your battery settings. You should be able to change cell target voltages.

niarbagem, 2017-01-13:
I have thought it's available only for Li-xx or not?

diggit, 2017-01-13:
Not sure, I don't have any charger nearby for next 2 days. Just try it.

niarbagem, 2017-01-13:
Of course l'll try, but only at the evening.

Next question:
Stavel, is it possible to make Pb charging strategy something that (taken from ctek msx 10)
[ctek msx 10](https://cloud.githubusercontent.com/assets/17687405/21926754/2d170fb0-d994-11e6-8e17-4347a28ebb03.jpg) ?

niarbagem, 2017-01-14:
Today i had tried to change Vcharge for Pb battery - it's unlimited ( i've seen 3,2xx per cell, and it want stops). 
So i'am ready to "16V" Pb battery :)

closed 2017-01-14 by niarbagem