stawel/cheali-charger #214: Is "Lipro Balance Charger" supported?

Harvie, 2017-02-19:
Can i use this with cheap chinese "Lipro Balance Charger" (fake imax b6)?
I can't see any label on chip, but it's hidden under lcd, so it's hard to see.
If so can you please add it to hardware list in readme?

[image]

stawel, 2017-02-19:
It's hard to say, they all look the same from outside, try to lift a bit your LCD and check your CPU
, or at least send a picture of your PCB.
If there is no label on your CPU then it's probably not supported.

Harvie, 2017-03-19:
I've made some photos and even managed to lift the lcd a bit... Can you tell more this way?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Harvie, 2017-03-19:
I've done little googling. And it might, just might be a **z51f3220**. I am not sure... Any luck with this MCU?

stawel, 2017-03-19:
Yes, it may be a z51f3220, unfortunately this CPU will not be supported,
I've tried to add it to the project but it's architektur is just to old.

Harvie, 2017-03-19:
It seems there are people on internet having same problem... I guess there's no lightweight cheali alternative targeting these devices yet?

stawel, 2017-03-20:
I'm not aware of any.

intruder777, 2017-07-31:
I have similar device (marked as 80W). Processor is labeled, but the label is scratched out with laser. However the first line can be clearly read as "ABOV". And people on the net say that second line is "MC96F6432".
Here it is:
http://abov.co.kr/en/index.php?Depth1=3&Depth2=1&Depth3=1&Depth4=1&Item=MC96F6432
http://aitendo3.sakura.ne.jp/aitendo_data/product_img/microcontroller/MC96F6432/MC96F6432_V1.2.pdf

Is it possible to port cheali-carger to this processor?

diggit, 2017-07-31:
Hi @intruder777,
This MCU is 8051 based and these are not supported by cheali. AFAIK there is no free multiplatform c++ compiler for 8051. Another problem could be only 1kiB ram. (atmega32 has 2kiB)
More here https://github.com/stawel/cheali-charger/issues/106#issuecomment-222520952

ossilator, 2026-09-28:
huh, the Z51F3220 datasheet is obviously only a slightly modified version of the MC96F6432 one, so it's the same thing.

anyway, the most realistic way to "support" this charger seems to be [replacing the MCU](https://github.com/martonmiklos/imaxclone_atmega) ...